gnuradio/pybombs #547: pybombs install 3.7.15 instead 3.8

mohmmadd, 2019-08-10:
When I try to install GNU Radio with these codes:
```
$ mkdir ~/prefix
$ pybombs prefix init -a default ~/prefix/default/ -R gnuradio-default
```
It installs 3.7.15 instead 3.8. 
What should I do to install 3.8?

velichkov, 2019-09-07:
Hi @mohmmadd, @ikarus23,

> When I try to install GNU Radio with these codes:
> 
> ```
> $ mkdir ~/prefix
> $ pybombs prefix init -a default ~/prefix/default/ -R gnuradio-default
> ```
> 
> It installs 3.7.13.5 instead 3.8.
> What should I do to install 3.8?

You need to do some modifications to the recipes and then install from gnuradio38 recipe. See https://github.com/gnuradio/gr-recipes/pull/155

ikarus23, 2019-09-07:
Ok, makes sense that this is a recipe issue. I think we can close this here and continue over at https://github.com/gnuradio/gr-recipes/.

mbr0wn, 2019-09-19:
You can `pybombs install gnuradio37`.

mbr0wn, 2019-09-19:
@mohmmadd to fix, run `pybombs recipes update`

closed 2019-09-19 by mbr0wn