gnuradio/pybombs #549: PyBOMBS.install_manager - ERROR - Package has no install method: autotools-bootstrap

Andreas237, 2019-09-10:
Greetings,

On Macbook, OSx Mojave 10.14.6 (18G95), x86_64 I am having issues with PyBOMBS.  I've read through the open issues, and searched the closed, but haven't found an answer.

The error is ```PyBOMBS.install_manager - ERROR - Package has no install method: autotools-bootstrap```

The log of the command and output is here:
```
me-MacBook-Pro:~ me$ pybombs prefix init ~/gnuradio -R gnuradio-default
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.16
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.prefix - WARNING - There already is a prefix in `/Users/me/gnuradio'.
Continue using this path Y/[N]? y
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.16
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.16
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO -
  - gnuradio
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: lxml in /usr/local/lib/python2.7/site-packages (4.4.1)
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting PyGTK
  Using cached https://files.pythonhosted.org/packages/7c/18/fa4f2de77500dd62a314fd845ff6e903ac2ce551164cb421c5750969f799/pygtk-2.24.0.tar.bz2
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: ********************************************************************
    * Building PyGTK using distutils is only supported on windows. *
    * To build PyGTK in a supported way, read the INSTALL file.    *
    ********************************************************************
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /private/tmp/pip-install-lhfBzz/PyGTK/
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Requirement already satisfied: requests in /usr/local/lib/python2.7/site-packages (2.22.0)
Requirement already satisfied: urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1 in /usr/local/lib/python2.7/site-packages (from requests) (1.25.3)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python2.7/site-packages (from requests) (2019.6.16)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/site-packages (from requests) (3.0.4)
Requirement already satisfied: idna<2.9,>=2.5 in /usr/local/lib/python2.7/site-packages (from requests) (2.8)
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
PyBOMBS.install_manager - ERROR - Package has no install method: autotools-bootstrap
```

I guess it's understandable that PyGTK fails since it's targeting Windows.  Any advice on how to proceed is welcome.

mbr0wn, 2019-09-19:
OK I think I know what's going on. On Mac OS X, we don't have a lot of ways to check if autotools and/or autoconf are installed. In fact, we only have one: The `cmd` package manager, which will run `autoconf` and `automake` and see if they exist. Barring that, PyBOMBS assumes that autotools are not installed, and it has no other way of installing it.

Since I can't reproduce this, here's a couple of questions:
- Do you know if automake and autoconf are installed?
- If you skip this test (by running `pybombs config --package autotools- bootstrap forceinstalled`), will it fail later or does it work?

mbr0wn, 2019-09-19:
...and if autotools are installed, are there better ways of knowing that it is?

Andreas237, 2019-10-16:
@mbr0wn Thank you for getting back to me!  I actually just upgrade Mac's recently, and got GRC running with no issues.

Sorry for the delayed response, and thank you for the in-depth look!

closed 2019-10-16 by Andreas237

saintofinternet, 2020-01-06:
i am facing this issue with Kali 2019.4 and still not able to solve it! 

i am running Kali on a VirtualBox and earlier it installed like a charm but now i get the same precise error...

`[ERROR] Package has no install method: python-click-plugins
`

joelunmsm2003, 2020-01-23:
Helpe me

pybombs install gr-gsm
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.17
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.install_manager - ERROR - Package has no install method: python-click-plugins

devnulling, 2020-01-23:
Try:
`sudo apt install python-click python-click-plugins `

If you use py3:
`sudo apt install python3-click python3-click-plugins`

SorinBaiuta, 2020-03-17:
Latest one did not worked 

apt install python-click-plugins
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package python-click-plugins