gnuradio/pybombs #555: [ERROR] Package has no install method: qt5

bairimenglan, 2019-09-28:
I have already installed qt package through Homebrew, but it still point out that there is no install method:qt5, how can i deal this case?

-------------------------------------------------------------------------------
```
~/SDR/gnuradio » pybombs prefix init ~/SDR/gnuradio -R gnuradio-default                                                                                                                qing@qingdeMacBook-Pro
[INFO] Prefix Python version is: 2.7.16
[INFO] PyBOMBS Version 2.3.4a0
[WARNING] There already is a prefix in `/Users/qing/SDR/gnuradio'.
Continue using this path Y/[N]? Y
[INFO] Prefix Python version is: 2.7.16
[INFO] Prefix Python version is: 2.7.16
[INFO] Installing default packages for prefix...
[INFO]
  - gnuradio
[INFO] Phase 1: Creating install tree and installing binary packages:
[ERROR] Package has no install method: qt5
```

brianredbeard, 2019-12-01:
This is specifically because an install type is being requested which the package `qt5` has no method/mechanism of supporting.  Specifically the recipe for QT5 [supports the following](
https://github.com/gnuradio/gr-recipes/blob/df3bc9a98ccaab14876e5684fa5e06699fb4410a/qt5.lwr#L21-L26):

```yaml
  deb: libqt5opengl5-dev && libqt5svg5-dev && qt5-default
  rpm: (qt5-qtbase-devel && qt5-qtsvg-devel) || (devel_qt5 && libqt5-qtsvg-devel)
  pacman: qt5-base
  port: qt5 && qt5-qtbase
  portage: dev-qt/qtgui
```

I experienced this same issue when attempting to first do a fully source based install, then when splitting the difference and using `pip` for some packages.

Be warned though, this problem is endemic across the various recipes.  Some other packages (e.g. [`python-click-plugins`](https://github.com/gnuradio/gr-recipes/blob/df3bc9a98ccaab14876e5684fa5e06699fb4410a/python-click-plugins.lwr#L25-L30)) only support binary installs _despite_ the fact that it's perfectly sufficient to do `pip` based installation.

robertpib, 2020-09-01:
got the same problem attempting to install gnuradio 3.8 using PyBOMBS in git console on win10 machine - see log:
```
$ pybombs auto-config
[INFO] Prefix Python version is: 3.7.5
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Using config file: None

user@MACHINE MINGW64 /c
$ pybombs recipes add-defaults
[INFO] Prefix Python version is: 3.7.5
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Creating new git cache in C:\Users\bert\.pybombs\gitcache
remote: Enumerating objects: 42, done.
remote: Counting objects: 100% (42/42), done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 1825 (delta 23), reused 5 (delta 2), pack-reused 1783
Receiving objects: 100% (1825/1825), 368.66 KiB | 1.61 MiB/s, done.
Resolving deltas: 100% (1483/1483), done.
From https://github.com/gnuradio/gr-recipes
 * [new branch]      homebrew   -> gr-recipes/homebrew
 * [new branch]      master     -> gr-recipes/master
Cloning into 'gr-recipes'...
warning: no common commits
remote: Enumerating objects: 40, done.
remote: Counting objects: 100% (40/40), done.
remote: Compressing objects: 100% (34/34), done.
remote: Total 519 (delta 23), reused 13 (delta 6), pack-reused 479
Receiving objects: 100% (519/519), 100.12 KiB | 1.05 MiB/s, done.
Resolving deltas: 100% (344/344), done.
From https://github.com/gnuradio/gr-etcetera
 * [new branch]      master     -> gr-etcetera/master
Cloning into 'gr-etcetera'...

user@MACHINE MINGW64 /c
$ pybombs prefix init ~/gnuradio -R gnuradio-default
[INFO] Prefix Python version is: 3.7.5
[INFO] PyBOMBS Version 2.3.4a0
[INFO] Creating directory `C:\Users\bert\gnuradio'
[INFO] Prefix Python version is: 3.7.5
[INFO] Creating new config file C:\Users\bert\gnuradio\.pybombs\config.yml
[INFO] Prefix Python version is: 3.7.5
[INFO] Installing default packages for prefix...
[INFO]
  - gnuradio
[INFO] Phase 1: Creating install tree and installing binary packages:
[ERROR] Package has no install method: qt5

user@MACHINE MINGW64 /c
$
```

ziyuanjun, 2021-08-05:
I got the same issue on Manjaro

> sudo pybombs install gnuradio
> [INFO] Prefix Python version is: 3.8.5
> [INFO] PyBOMBS Version 2.3.4
> [INFO] Phase 1: Creating install tree and installing binary packages:
> [ERROR] Package has no install method: qt5

closed 2021-08-28 by willcode

k-bx, 2022-06-03:
Why was this closed? Still getting this error on my macOS:

```
$ pybombs prefix init ~/tmp/gnuradio -R gnuradio-default
[INFO] Prefix Python version is: 3.9.13
[INFO] PyBOMBS Version 2.3.5
[WARNING] There already is a prefix in `/Users/kb/tmp/gnuradio'.
Continue using this path Y/[N]? Y
[INFO] Prefix Python version is: 3.9.13
[INFO] Prefix Python version is: 3.9.13
[INFO] Installing default packages for prefix...
[INFO]
  - gnuradio
[INFO] Phase 1: Creating install tree and installing binary packages:
[ERROR] Package has no install method: qt5

```

mhenstell, 2022-06-15:
Same error on 11.6.7

k-bx, 2022-06-16:
Created a https://github.com/gnuradio/pybombs/issues/618 for macOS specific report that is still "open"