gnuradio/pybombs #556: Cannot install 3.7 after attempting 3.8 via Pybombs

cclamb, 2019-09-28:
Attempting to install gnuradio 3.7 after installing 3.8 via P$ ybombs:
$ pybombs prefix init ~/gnuradio_3.7 -a gr37
$ pybombs -p gr37 -vv install gnuradio37

Results in:
PyBOMBS.Fetcher.git - DEBUG - Requirements met.
PyBOMBS.Fetcher.git - DEBUG - We have git version 2.20.1
PyBOMBS.Fetcher.git - DEBUG - Using url - https://github.com/gnuradio/gnuradio.git                                                       
PyBOMBS.Fetcher.git - DEBUG - Adding remote into git ref
PyBOMBS.GitCacheManager - DEBUG - Adding remote: gnuradio37 -> https://github.com/gnuradio/gnuradio.git                                  
PyBOMBS.GitCacheManager - DEBUG - Remote URL https://github.com/gnuradio/gnuradio.git already registered.                                
fatal: No such remote or remote group: gnuradio37
PyBOMBS.Fetcher - ERROR - Unexpected error while fetching git+https://github.com/gnuradio/gnuradio.git.                                  
PyBOMBS.Fetcher - ERROR - Command '['git', 'remote', 'update', 'gnuradio37']' returned non-zero exit status 1.                           
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gnuradio37:                                                    
Unable to fetch recipe gnuradio37
PyBOMBS.install_manager - ERROR - Error installing package gnuradio37. Aborting.                                                         
(cr) cclamb@mordor:~$

marcusmueller, 2019-09-28:
transferring to pybombs issue tracker.

DoHaiSon, 2019-09-29:
The same error too, i want install GRC 3.7 and followed : [https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux](url) . But it installed GRC 3.8 and i can't install iqbal, gqrx,... 
So i delete 3.8: 
`pybombs remove gnuradio`
Then install new 3.7:
`pybombs install gnuradio37`

But Error T.T . Anyone help me, how to install BladeRF, GnuRadio, Gqrx and WxGui block in GRC. I'm using ubuntu 18.04 LTS. Thanks so much!!