gnuradio/pybombs #561: Fresh Ubuntu 18.04 LTS Problems (orc-0.4, gmp, mpir, thrift, numpy)

catalinalb, 2019-12-01:
Hello, 
Here are the unsolved problmes found: 

Checking for module 'orc-0.4 > 0.4.11'
CPU missing xgetbv, Overruled arch avx
Checking for module 'gmp'
Checking for module 'mpir >= 3.0'
Checking for module 'thrift'
Python checking for numpy - not found

installing gmp with over pybombs dose not solve the problem. Any ideas?

Here are the steps that I used so far:
sudo apt-get update
sudo apt-get install git libyaml-dev libssl-dev python-dev
sudo apt-get install texinfo
mkdir sdr
cd sdr
sudo apt-get install python-pip
sudo easy_install pip (sudo pip
sudo pip install PyBOMBS
sudo pip install --upgrade git+https://github.com/gnuradio/pybombs.git
pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
pybombs prefix init ~/sdr -a myprefix
pybombs install gnuradio gr-osmosdr

Thank you,
Catalin

velichkov, 2019-12-03:
Hi @catalinalb,

> Python checking for numpy - not found

> sudo pip install PyBOMBS

My advice is to reinstall pybombs using `pip3`

```
pip uninstall pybombs
pip3 install --upgrade git+https://github.com/gnuradio/pybombs.git
```

This way pybombs would install the python3 dependencies, e.g. it would install `python3-numpy` instead of `python-numpy`

An alternative solution is to set `python_ver: 3.6` in `~/.pybombs/config.yml` (change `3.6` with the python3 version installed on your system). See https://github.com/gnuradio/pybombs/#python-version

catalinalb, 2019-12-03:
Thank you for your answer. 
I have just done that but... uninstall and install back and now the pybombs reply is: command not found :-1:

velichkov, 2019-12-03:
Could you give us the exact commands you've executed including all arguments?

catalinalb, 2019-12-05:
Hello,
So I just reinstalled ubuntu 18.07 LTS. (again)
Here are the commands that I ran:
sudo apt update 
sudo apt dist-upgrade
sudo apt upgrade
sudo apt install screen net-tools putty synaptic git
I did manually install OpenCL from intel.com
sudo apt-get install libyaml-dev libssl-dev python-dev
mkdir sdr
cd sdr
sudo apt-get install python-pip
sudo easy_install pip --- responded with error
sudo apt install python python-pip (allready installed)
sudo pip install --upgrade git+https://github.com/gnuradio/pybombs.git (I did run this to update)
sudo pip install PyBOMBS
pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
pybombs prefix init ~/sdr -a gnuradio-stable (yes I did choose gnuradio-stable)
pybombs install gnuradio gr-osmosdr

and then it started to work ... hopefully this time everything will run without problems .... so I hoped...
and then after finishing installing uhd and gnuradio I got the same errors:

Submodule path 'volk': checked out '1299d72c396a88fd2679adfd7a919ac00d2cf678'
Configuring: (100%) [===============================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- User set python executable /usr/bin/python3.6
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.9", minimum required is "2.7.6") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.9", minimum required is "2.7.6") 
-- Python checking for six - python 2 and 3 compatibility library - found
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- 
-- Configuring testing-support support...
--   Dependency Boost_FOUND = 1
--   Enabling testing-support support.
[... 115 lines cut ...]
-- Python checking for numpy - not found
-- 
-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = TRUE
--   Dependency PYYAML_FOUND = TRUE
--   Dependency MAKO_FOUND = TRUE
--   Dependency PYGI_FOUND = TRUE
--   Dependency GTK_GI_FOUND = TRUE
--   Dependency CAIRO_GI_FOUND = TRUE
--   Dependency PANGOCAIRO_GI_FOUND = TRUE
--   Dependency NUMPY_FOUND = FALSE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:95 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/ctl/sdr/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/ctl/sdr/src/gnuradio/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gnuradio:
Configuration failed
[ERROR] Error installing package gnuradio. Aborting.

catalinalb, 2019-12-05:
Loading version 3.8.0.0 into constants...
-- Checking for module 'thrift'

Is it still installing the version 3.8....? I did choose gnuradio-stable so It should install 3.7 ....!

velichkov, 2019-12-05:
Hi @catalinalb,

> So I just reinstalled ubuntu 18.07 LTS. (again)

My advice is to first try installing it in a virtual machine or a docker container and only when the installation is successful attempt installation on your workstation. You could take snapshots of the VM that you could easily revert to instead of reinstalling the whole OS.

> pybombs prefix init ~/sdr -a gnuradio-stable (yes I did choose gnuradio-stable)

That's not correct, it should be `-R` and not `-a`. The `-a` is for `alias` while `-R` is for `recipe`
```
pybombs prefix init ~/sdr -R gnuradio-stable
```
This command should start installing `gnuradio-3.7` and all its dependencies.

See also https://github.com/gnuradio/pybombs/#to-run-pybombs-quickstart

> -- Dependency NUMPY_FOUND = FALSE

If for some reason it still fails with this then install `python-numpy` manually.

> -- Checking for module 'thrift'

In the `gnuradio-stable.lwr` recipe `trifth` is marked as an optional dependencies so if you don't need it you should not worry about the `thrift` related messages.

> Is it still installing the version 3.8....? I did choose gnuradio-stable so It should install 3.7 ....!

It sill tries to install `3.8` as you did not init your prefix correctly.

catalinalb, 2019-12-06:
Thank you very much for the clarification.
Catalin

Kimiar777, 2020-04-01:
when i try to install gr-gsm using pybombs i am getting this error

root@kali:~# pybombs install gr-gsm
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 3.7.7     
PyBOMBS - INFO - PyBOMBS Version 2.3.3                             
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:                                       
Install tree:                                                      
|
\- gr-gsm
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:                                           
PyBOMBS.install_manager - INFO - Installing package: gr-gsm
PyBOMBS.Packager.source - WARNING - Build dir already exists: /usr/local/src/gr-gsm/build                                             
Configuring: (100%) [============================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.                                            
-- Extracting version information from git describe...
-- Found Boost: /usr/include (found suitable version "1.67.0", minimum required is "1.35") found components: filesystem system thread chrono date_time atomic 
-- 
-- Checking for module SWIG
-- Found SWIG version 4.0.1.
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found Boost: /usr/include (found suitable version "1.67.0", minimum required is "1.67.0") found components: date_time program_options filesystem system regex thread unit_test_framework chrono atomic 
CMake Warning at CMakeLists.txt:135 (find_package):
  Found package configuration file:

    /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake

  but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
  be NOT FOUND.  Reason given by package:

  Gnuradio could not be found because dependency Volk could not be found.

CMake Error at CMakeLists.txt:150 (message):
  GnuRadio Runtime required to compile gr-gsm

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/gr-gsm/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/gr-gsm/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.                                                  
PyBOMBS.Packager.source - ERROR - Problem occurred while building package gr-gsm:                                                     
Configuration failed                                               
PyBOMBS.install_manager - ERROR - Error installing package gr-gsm. Aborting.

velichkov, 2020-04-01:
Hi @Kimiar777,

> when i try to install gr-gsm using pybombs i am getting this error
>```
> ## root@kali:~# pybombs install gr-gsm
> CMake Warning at CMakeLists.txt:135 (find_package):
> Found package configuration file:
> 
> 
> /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake
> 
> 
> but it set Gnuradio_FOUND to FALSE so package "Gnuradio" is considered to
> be NOT FOUND. Reason given by package:

For now you need to install gr-gsm from the [porting_to_gr38](https://github.com/ptrkrysik/gr-gsm/tree/porting_to_gr38 ) branch or from the [maint-3.8](https://github.com/velichkov/gr-gsm/tree/maint-3.8) branch from my fork as the master branch does not yet support gnuradio 3.8.
```
pybombs config -P gr-gsm gitbranch porting_to_gr38
pybombs refetch gr-gsm
pybombs install gr-gsm
```
See also https://github.com/ptrkrysik/gr-gsm/issues/475 and https://github.com/ptrkrysik/gr-gsm/issues/480

zkmartin, 2020-06-15:
Hi, @velichkov!
I installed the gnuradio 3.8 successfully with pip3. Then I wanted to install the gr-osmosdr,gr-mapper,gr-mediatools，but I did not installed successfully.Are these resipes are not support fro gnuradio 3.8 yet? Should I install gnuradio 3.7 with pip3 following the steps above？ And  how to install those recipes?

velichkov, 2020-06-15:
Hi @zkmartin,

> Then I wanted to install the gr-osmosdr,gr-mapper,gr-mediatools，but I did not installed successfully.Are these recipes are not support for gnuradio 3.8 yet? 

`gr-osmosdr` supports gnuradio 3.8, I don't about the other two.

> Should I install gnuradio 3.7 with pip3 following the steps above？

No, gnuradio 3.7 does not support python3.

> And how to install those recipes?

When reporting errors it's  really important to provide detailed information - the exact error messages, build logs, etc...

zkmartin, 2020-06-16:
> Hi @zkmartin,
>
>
> `gr-osmosdr` supports gnuradio 3.8, I don't about the other two.
>
>
> No, gnuradio 3.7 does not support python3.
>
>
> When reporting errors it's really important to provide detailed [...]

Hi, @velichkov!
Thanks so much for your help！
I try to install gnuradio3.8 with pip3 and gnuradio 3.7 with pip2. Each of them is on the fresh unbuntu 18.04.

### 1.gnuradio3.8 with pip3(successfully but gr-mediatools installation failed)

sudo apt-get install python3-pip
pip3 install --upgrade pip
pip3 install git+https://github.com/gnuradio/pybombs.git
pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
pybombs prefix init /usr/local -a myprefix -R gnuradio-default

pybombs install osmo-sdr rtl-sdr gnuradio hackrf bladeRF airspy gr-iqbal libosmo-dsp gr-osmosdr gqrx
pybombs gnss-sdr
cd /usr/local
. ./setup_env.sh
gnuradio-config-info -v 
gnuradio version 3.8.1.0

Then pybombs install gr-mediatools.It 
Install tree:
|
\- gr-mediatools
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gr-mediatools
[WARNING] Build dir already exists: /usr/local/src/gr-mediatools/build
Configuring: (100%) [=========================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
-- Boost version: 1.65.1
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
CMake Error at CMakeLists.txt:94 (message):
  GnuRadio Runtime required to compile mediatools

And I installed the gmp and mpir with the package from official website using the following commands:
$tar -jvxf gmp-6.1.2.tar.bz2
$cd gmp-6.1.2
$./configure --enable-cxx
$make
$make check
$sudo make install

But the report still shows that no package ‘gmp found and no package ‘mpir’ found。
What should I do with this？’

---------------------------------------------------------------------------------------------------------
### 2. gnuradio3.7 with pip2
sudo apt-get install python
sudo apt-get install python-pip
pip install --upgrade pip
pip install git+https://github.com/gnuradio/pybombs.git
pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
pybombs prefix init /usr/local -a myprefix -R gnuradio-stable
[... 336 lines cut ...]
  File "/usr/lib/python3.6/traceback.py", line 3, in <module>
    import collections
  File "/usr/lib/python3.6/collections/__init__.py", line 32, in <module>
    from reprlib import recursive_repr as _recursive_repr
  File "/usr/local/lib/python2.7/dist-packages/reprlib/__init__.py", line 7, in <module>
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.
-- Python checking for numpy 1.7 or greater - "import numpy" failed
-- 
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = 1
--   Dependency HAVE_PYTHON_PLAT_MIN_VERSION = FALSE
--   Dependency HAVE_PYTHON_MODULE_MAKO = FALSE
CMake Error at cmake/Modules/UHDComponent.cmake:59 (message):
  Dependencies for required component LibUHD not met.
Call Stack (most recent call first):
  CMakeLists.txt:363 (LIBUHD_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/usr/local/src/uhd/host/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/src/uhd/host/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package uhd:
Configuration failed
[ERROR] Error installing package uhd. Aborting.

velichkov, 2020-06-16:
Hi @zkmartin,

> pybombs prefix init /usr/local -a myprefix -R gnuradio-default

My advice is to avoid installing in `/usr/local/`, use a sub-directory in your home directory instead. This way if something goes wrong you can just delete the whole directory and start from scratch.

> Then pybombs install gr-mediatools.It
> Install tree:
>...........
> CMake Error at CMakeLists.txt:94 (message):
> GnuRadio Runtime required to compile mediatools

It looks like `gr-mediatools` has not been ported to 3.8 yet as the last commit is from 2016. If you want to port it read [GNU Radio 3.8 OOT Module Porting Guide](https://wiki.gnuradio.org/index.php/GNU_Radio_3.8_OOT_Module_Porting_Guide).

> And I installed the gmp and mpir with the package from official website using the following commands:
> $tar -jvxf gmp-6.1.2.tar.bz2
> $cd gmp-6.1.2
> $./configure --enable-cxx
> $make
> $make check
> $sudo [...]

You can install `gmp` from the Ubuntu's repositories - `sudo apt-get install libgmp-dev`. You may also consider uninstalling the version installed from source, go in the build directory and execute - `sudo make uninstall` as having multiple versions could cause various problems.

> ### 2. gnuradio3.7 with pip2
>
> sudo apt-get install python
> sudo apt-get install python-pip
> pip install --upgrade pip
> pip install git+https://github.com/gnuradio/pybombs.git
> pybombs recipes add [...]

Use a different prefix (e.g not `/usr/local/`) then the one you've used to install gnuradio 3.8 in. If you want two gnuradio versions my advice is to create two sub-directories in your home directory - for example `~/gr38` and `~/gr37` and use these instead of `/usr/local/`.

> File "/usr/lib/python3.6/collections/**init**.py", line 32, in
> from reprlib import recursive_repr as _recursive_repr
> File "/usr/local/lib/python2.7/dist-packages/reprlib/**init**.py", line 7, in
> raise ImportError('This package should not be accessible on Python 3. '
> ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.

Do not reuse the same terminal in which you've run the `./setup_env.sh` script from the gnuradio 3.8 prefix. Open a new terminal and check that `PYTHONPATH` environment variable _does not_ contain python3 paths.

zkmartin, 2020-06-17:
> Hi @zkmartin,
>
>
> My advice is to avoid installing in `/usr/local/`, use a sub-directory in your home directory instead. This way if something goes wrong you can just delete the whole directory and [...]

Hi velichkov!
Thanks for your help!

I try to open a new terminal and check that PYTHONPATH environment variable does not contain python3 paths. This works  but new error occurrs.

open a new terminal
export PYTHONPATH="/usr/bin/python2.7:$PYTHONPATH"

sudo apt-get install python-pip
pip install --upgrade pip
pip install git+https://github.com/gnuradio/pybombs.git
pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git
pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git
pybombs prefix init /home/test -a myprefix -R gnuradio-stable

-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = TRUE
--   Dependency CHEETAH_FOUND = FALSE
--   Dependency LXML_FOUND = TRUE
--   Dependency PYGTK_FOUND = TRUE
--   Dependency NUMPY_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:45 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/test/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/test/src/gnuradio/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gnuradio:
Configuration failed
[ERROR] Error installing package gnuradio. Aborting.

What does this mean?

velichkov, 2020-06-17:
Hi @zkmartin,

> -- Dependency CHEETAH_FOUND = FALSE

You need to install cheetah - `sudo apt-get install python-cheetah`

You can easily install the latest gnuradio 3.7.* version from the [gnuradio-releases-3.7](https://launchpad.net/~gnuradio/+archive/ubuntu/gnuradio-releases-3.7) PPA repository.
```
sudo add-apt-repository ppa:gnuradio/gnuradio-releases-3.7
sudo apt-get update
sudo apt-get install gnuradio-dev
```
You can install 3.8 from [gnuradio-releases](https://launchpad.net/~gnuradio/+archive/ubuntu/gnuradio-releases) PPA repository.

Also note that if you choose to install from the PPA repositories you can't install two different gnuradio versions at the same time.

zkmartin, 2020-06-19:
> Hi @zkmartin,
>
>
> You need to install cheetah - `sudo apt-get install python-cheetah`
>
> You can easily install the latest gnuradio 3.7.* version from the [...]

> 
> Also note that if you choose to install from the PPA repositories you can't install two different gnuradio versions at the same time.

Hi velichkov!
Thanks for your help! I'm sorry to bother you so many times.

I installed gnuradio3.7 with pybombs successfully with your intructions.
When I excecute the following command to install some recipes,
pybombs install osmo-sdr rtl-sdr hackrf bladeRF airspy gr-iqbal libosmo-dsp gr-osmosdr gqrx
Errors occurred.
Cmake Error at CMakeLists.txt:45(find_package):
Could not find a configuration file for package "Gnuradio" that is
compatible with requested version "3.8".
The following configuration files were considered but not accepted:
/usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.7.14.0

My target is to install gr-mediatools and gr-mapper and these recipes only support gnuradio3.7.
So I want to install gnuradio3.7.
Is there a method to use the old version gr-recipes?

Thanks a lot!

zkmartin, 2020-06-23:
> Hi @zkmartin,
>
>
> You need to install cheetah - `sudo apt-get install python-cheetah`
>
> You can easily install the latest gnuradio 3.7.* version from the [...]

Hi，velichkov！
Thanks for your help! 
I installed gnuradio3.7 successfully with pip2 on fresh ubuntu16.04.
I used pybombs to install gr-osmodr but It seemed that the default gr-osmosdr version is not supported to gnuradio3.7.
So I  used git clone -b gr3.7 git://git.osmocom.org/gr-osmosdr to clone the gr-osmosdr to local.
And I installed it.
Errors occurred.
[  2%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o
In file included from /usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.h:26:0,
                 from /usr/local/src/gr-osmosdr/lib/source_impl.cc:72:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.h:291:50: error: ‘>>’ should be ‘> >’ within a nested template argument list
   static std::list<boost::weak_ptr<struct bladerf>> _devs;  /**< dev cache */
                                                  ^
In file included from /usr/local/src/gr-osmosdr/lib/source_impl.cc:72:0:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.h:141:32: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
   const float SCALING_FACTOR = 2048.0f;
                                ^
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:62: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o] Error 1
CMakeFiles/Makefile2:140: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package gr-osmosdr:
Build failed.
[ERROR] Error installing package gr-osmosdr. Aborting.

How should I deal with this.It seems that the compiling environment is not support to the files.

And I try the checkout to the version that commited in  2017.
Different errors occurred.
[  2%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_source_c.cc.o
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.cc: In member function ‘virtual double bladerf_source_c::get_center_freq(size_t)’:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.cc:250:69: error: cannot convert ‘uint32_t* {aka unsigned int*}’ to ‘bladerf_frequency* {aka long unsigned int*}’ for argument ‘3’ to ‘int bladerf_get_frequency(bladerf*, bladerf_channel, bladerf_frequency*)’
   ret = bladerf_get_frequency( _dev.get(), BLADERF_MODULE_RX, &freq );
                                                                     ^
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:257: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_source_c.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_source_c.cc.o] Error 1
CMakeFiles/Makefile2:140: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package gr-osmosdr:
Build failed.
[ERROR] Error installing package gr-osmosdr. Aborting.

zkmartin, 2020-06-23:
> Hi @zkmartin,
>
>
> You need to install cheetah - `sudo apt-get install python-cheetah`
>
> You can easily install the latest gnuradio 3.7.* version from the [...]

Hi，@velichkov！ 
Thanks for your help! I'm sorry to bother you so many times.
I installed gnuradio3.7 successfully with pip2 on fresh ubuntu16.04.
I used pybombs to install gr-osmodr but It seemed that the default gr-osmosdr version is not supported to gnuradio3.7.
So I  used git clone -b gr3.7 git://git.osmocom.org/gr-osmosdr to clone the gr-osmosdr to local.
And I installed it.
Errors occurred.
[  2%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o
In file included from /usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.h:26:0,
                 from /usr/local/src/gr-osmosdr/lib/source_impl.cc:72:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_common.h:291:50: error: ‘>>’ should be ‘> >’ within a nested template argument list
   static std::list<boost::weak_ptr<struct bladerf>> _devs;  /**< dev cache */
                                                  ^
In file included from /usr/local/src/gr-osmosdr/lib/source_impl.cc:72:0:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.h:141:32: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
   const float SCALING_FACTOR = 2048.0f;
                                ^
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:62: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/source_impl.cc.o] Error 1
CMakeFiles/Makefile2:140: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package gr-osmosdr:
Build failed.
[ERROR] Error installing package gr-osmosdr. Aborting.

How should I deal with this.It seems that the compiling environment is not support to the files.

And I also tried the checkout to the version that commited in  2017.
Different errors occurred.
[  2%] Building CXX object lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_source_c.cc.o
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.cc: In member function ‘virtual double bladerf_source_c::get_center_freq(size_t)’:
/usr/local/src/gr-osmosdr/lib/bladerf/bladerf_source_c.cc:250:69: error: cannot convert ‘uint32_t* {aka unsigned int*}’ to ‘bladerf_frequency* {aka long unsigned int*}’ for argument ‘3’ to ‘int bladerf_get_frequency(bladerf*, bladerf_channel, bladerf_frequency*)’
   ret = bladerf_get_frequency( _dev.get(), BLADERF_MODULE_RX, &freq );
                                                                     ^
lib/CMakeFiles/gnuradio-osmosdr.dir/build.make:257: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_source_c.cc.o' failed
make[2]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/bladerf/bladerf_source_c.cc.o] Error 1
CMakeFiles/Makefile2:140: recipe for target 'lib/CMakeFiles/gnuradio-osmosdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-osmosdr.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package gr-osmosdr:
Build failed.
[ERROR] Error installing package gr-osmosdr. Aborting.

velichkov, 2020-06-24:
Hi @zkmartin,

> I installed gnuradio3.7 successfully with pip2 on fresh ubuntu16.04.

Why did you chose to install gnuradio 3.7 from source using pybombs on ubuntu 16.04/18.04?

Compiling old versions of software on old distributions is a tricky business and most of the time it does not worth the efforts.  My advice is to post a message to the [mailing list](https://wiki.gnuradio.org/index.php/MailingLists) or [IRC/matrix chat](https://wiki.gnuradio.org/index.php/IRC) and describe what do you need gnuradio for, what SDR hardware do you have, do you need some OOT modules, etc... Much more people are reading the mailing list and the chances to receive advice are better there.

zkmartin, 2020-07-03:
> Hi @zkmartin,
>
>
> Why did you chose to install gnuradio 3.7 from source using pybombs on ubuntu 16.04/18.04?
>
> Compiling old versions of software on old distributions is a tricky business and most of [...]

I installed successfully! Thank you so much for your help!