gnuradio/pybombs #562: Libzmq (Ubuntu 18.04 LTS)

catalinalb, 2019-12-03:
Hello and sorry for the long error list but I don#t know what should I do next. I have ubuntu 18.04 LTS and I am trying to install gnuradio 3.7 so I can have acess to gr-fosphor. UNtill now I did installed pybombs and pybombsgui I did have the pybombs upgraded. I also created a list with all the dependencies that I installed until now:

sudo apt-get install git libyaml-dev libssl-dev python-dev
sudo apt-get install texinfo python-pip python3 ipython python-matplotlib python-ipython python-scipy python-numpy python-pip python-wxgtk3.0 multimon sox cmake xorg-dev libglu1-mesa-dev nvidia-opencl-dev opencl-headers nvidia-modprobe opencl-headers libboost-system-dev libboost-thread-dev python3-numpy python3-psutil python3-zmq python3-pyqt5 g++ libpython3-dev python3-pip cython3 libfreetype6-dev ocl-icd-opencl-dev python-opengl lsb-core libglib2.0-dev libgtk2.0-dev libgtkdatabox-dev libmatio-dev libfftw3-dev libxml2 libxml2-dev bison flex libavahi-common-dev libavahi-client-dev libcurl4-openssl-dev libjansson-dev cmake libaio-dev python3-pip qt5-default python-pyqt5 libyaml-0-2 libyaml-dev lua-yaml lua-yaml-dev python-yaml python3-yaml python3-yaml-dbg libqwt5-qt4 libqwt5-qt4-dev libcanberra-gtk0 yasm
sudo apt-get install -y libsctp-dev gawc

I have manually installed Boost to the latest version since it was requierd. 
I did install orc 0.4 from github
I did install gmc and mpir still it seems that they are still unreachable by the pybombs.
I did install cppzmq and libzmq from github
I also installed all the dependencies for gnuradio 3.7 and 3.8 for ubuntu 18.04 from the gnuradio installation instructions maybe this wasn#t the right thing to do...???

Thing is pybombs is installing under gnuradio the latest version 3.8 when in fact I need 3.7....

still now the libzmq problem still persists: Please help!!! 

-- Generating done
-- Build files have been written to: /home/ctl/sdr/src/gnuradio/build
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gnuradio:
Configuration failed
[ERROR] Error installing package gnuradio. Aborting.
[ERROR] Install Failed
[INFO] Install python-apt to speed up apt processing.
[INFO] Install python-apt to speed up apt processing.
[INFO] Preparing SDK packages
[INFO] Loading SDK packages - successful !
[INFO] Preparing application and baseline packages
[INFO] Loading application and baseline packages - Successful !
[INFO] Install python-apt to speed up apt processing.
[INFO] Final packages - [[], [u'gr-foo'], []] ['update', 'install', 'remove']
[INFO] Install python-apt to speed up apt processing.
[INFO] Preparing gr-foo for installation
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gr-foo
   |
   \- gnuradio
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gnuradio
[WARNING] Build dir already exists: /home/ctl/sdr/src/gnuradio/build
Configuring: (100%) [=========================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  cmake/Modules/GrBoost.cmake:72 (find_package)
  CMakeLists.txt:354 (include)

CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  cmake/Modules/GrBoost.cmake:72 (find_package)
[... 3860 lines cut ...]
CMake Error at cmake/Modules/UseSWIG.cmake:573 (add_library):
  Target "video_sdl_swig" links to target "Boost::regex" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/Modules/GrSwig.cmake:133 (swig_add_library)
  gr-video-sdl/swig/CMakeLists.txt:38 (GR_SWIG_MAKE)

CMake Error at cmake/Modules/UseSWIG.cmake:573 (add_library):
  Target "video_sdl_swig" links to target "Boost::thread" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
Call Stack (most recent call first):
  cmake/Modules/GrSwig.cmake:133 (swig_add_library)
  gr-video-sdl/swig/CMakeLists.txt:38 (GR_SWIG_MAKE)

CMake Error at gr-vocoder/lib/CMakeLists.txt:23 (add_library):
  Target "gnuradio-vocoder" links to target "Boost::thread" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at gr-vocoder/lib/CMakeLists.txt:23 (add_library):
  Target "gnuradio-vocoder" links to target "Boost::program_options" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED

velichkov, 2019-12-03:
Hi @catalinalb,

> I have ubuntu 18.04 LTS and I am trying to install gnuradio 3.7 so I can have acess to gr-fosphor.

Right now the easiest way is to install `gr-fosphor` from the Ubuntu's [apt repository](https://packages.ubuntu.com/bionic/gr-fosphor)

```
sudo apt-get install gr-fosphor
```

>  UNtill now I did installed pybombs and pybombsgui I did have the pybombs upgraded. 

In my opinion the pybombs installation method is harder then apt-get and it's worth it only if you want to have the latest and greatest versions.

> Thing is pybombs is installing under gnuradio the latest version 3.8 when in fact I need 3.7....

For `3.7` you need to init your prefix with `gnuradio-stable` recipe instead of `gnuradio-default`. See https://github.com/gnuradio/pybombs/#recipes

catalinalb, 2019-12-03:
Thank you again for the fast valuable answer. I'm using the "Chapter 2 installation guide from Field Experiment SDR.
I did manage to win the fight a few times with installing it but it didn't last long untill I messed it up.

I was able to have the IEEE802.11 package installed And all the options especially the ones by Bastian Bloessl.

Thank you for the tip with the prefix.

If in install it over synaptic it won't work since I need some QT stuff and OpenCL for the Fosphor. You can see my YouTube video with Fosphor and I am playing to do more although they are a boring without my voice on them...
https://youtu.be/DzHIJ1GP8_s

Thanks again for your help.

If I manage to master this installation I will probably write some instructions somewhere.

Bye

catalinalb, 2019-12-04:
Hi,
Just one more question:
Should I use the pip command for installing pybombs for the gnuradio-stable version ? or I can use pip3 command ? 

I know that gnuradio 3.7 (stable) is using python2.7 so I am not sure if I should use pip or pip3

Thank you

velichkov, 2019-12-04:
> Thank you again for the fast valuable answer. I'm using the "Chapter 2 installation guide from Field Experiment SDR.

Is [this](https://fieldxp.com/Chapter-2-Installing-the-Software.pdf) the guide? As it usually happens with such tutorials sooner or later they became outdated.

> If in install it over synaptic it won't work since I need some QT stuff and OpenCL for the Fosphor. 

OK. You could also try installing everything from the apt repository except gr-fosphor and then modify and rebuild gr-fosphor only.

> Should I use the pip command for installing pybombs for the gnuradio-stable version ? or I can use pip3 command ?
>
> I know that gnuradio 3.7 (stable) is using python2.7 so I am not sure if I should use pip or pip3

In your case better to use `pip`