gnuradio/pybombs #563: Ubuntu 20.04 LTS gr-iqbal Configuration Failed problem with 3.8

catalinalb, 2019-12-06:
Hello,

So I am running Ubuntu 20.04 LTS and I did install PyBombs with:
sudo pip uninstall pybombs
sudo pip3 install --upgrade git+https://github.com/gnuradio/pybombs.git
pybombs auto-config
pybombs recipes add-defaults
pybombs prefix init ~/sdr -R gnuradio-stable

went with the problem of qwt6 but I solved it by repalcing && with || into gr-recipes/qwt6.lwr

run it again and it finished without errors. Still even after running the source command the gnuradio is not starting because of the LD_LIBRARY_PATH

then I tried: pybombs install gr-osmosdr and it hanged with the following error: 

nstall tree:
|
\- gr-osmosdr
   |
   \- gr-iqbal
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: gr-iqbal
[WARNING] Build dir already exists: /home/ctl/sdr/src/gr-iqbal/build
Configuring: (100%) [=================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
CMake Error at CMakeLists.txt:40 (find_package):
  Could not find a configuration file for package "Gnuradio" that is
  compatible with requested version "3.8".

  The following configuration files were considered but not accepted:

    /home/ctl/sdr/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.7.13.5

-- Configuring incomplete, errors occurred!
See also "/home/ctl/sdr/src/gr-iqbal/build/CMakeFiles/CMakeOutput.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gr-iqbal:
Configuration failed
[ERROR] Error installing package gr-iqbal. Aborting.

I did choose the pip3 command for installing  pybombs because with the pip command it gives me an error with python-clicks.... 

Thank you

velichkov, 2019-12-10:
Hi @catalinalb,

> So I am running Ubuntu 20.04 LTS 

Why did you switch to Ubuntu 20.04 LTS? This version is still work in progress and will be released next year.

> \- gr-osmosdr
>    |
>    \- gr-iqbal
> [INFO] Phase 1 complete: All binary dependencies installed.
> [INFO] Phase 2: Recursively installing source packages to prefix:
> [INFO] Installing package: gr-iqbal [...]

The `gr-iqbal` was just ported to gnuradio `3.8` and its `master` branch now contains a version for `3.8` while the version for `3.7` is in the `gr3.7` branch.

You need to either edit the `.pybombs/recipes/gr-recipes/gr-iqbal.lwr` recipe and change `gitbranch:` value to `gr3.7` or execute the following.
```
pybombs config -P gr-iqbal gitbranch gr3.7
```
Note that `gr-osmosdr` is still being ported and its `master` branch still contains version for `3.7` but it could be switched at any time so its probably better to do the same change for gr-osmosdr as well
```
pybombs config -P gr-osmosdr gitbranch gr3.7
```
and then you need to refetch these versions and then rebuild.
```
pybombs refetch gr-iqbal gr-osmosdr
pybombs install gr-osmosdr
```

catalinalb, 2019-12-21:
Hello
The thing is I only want gnuradio 3.7 because it runs great with gr-gsm, with modules from Bastian and especially gr-fosphor.
Is there a way of adding only the recipes for gr3.7? I thought  I can replace the command

pybombs recipes add-default with pybombs recipes  add-gr3.7 but it didn't work out

For a couple of months, I am still fighting with the pybombs installation. It worked a few times following the instructions from Bastian with instant-gnuradio combined with pybombsgui.

But in recent months I just can't get it to work. Ubuntu 18.04 LTS Ubuntu 19.10, Kali.... 
I am using the pybombsgui so I can see all the modules that exist.
I followed installation instructions but as you said some are running obsolete after a while.

Since I have some free time now I would like to get it to work in a clean environment. Ubuntu 19.10 and 20.04 have the nvidia drivers installed so I think it's easier to enable CUDA there as it is in ubuntu 18.04. 

Thank you
Catalin

cyrus104, 2020-04-09:
@catalinalb, it's been a bit since you were testing this. I wanted to see if you had made any progress with pybombs/gnuradio in Ubuntu 20.04? I know it's still in beta but we are starting to move all of our projects over to the new LTS, we expect a lot of testing will be needed and need to start working through the issues.

Did you run into any issues with pygtk when you ran through the install?

rfguru-rufus, 2021-06-15:
@velichkov now the package works for Ubuntu 20.04 LTS 64 bit? It has been while this issue is creating problem with Ubuntu 20.04