gnuradio/pybombs #567: Failed to install qt5 - Ubuntu 18.04.3

Edson-PU2TNO, 2020-01-09:
Is there any fix for this?

Install tree:
|
\- qt5
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[TRACE] qt5 has cached installed-status: False
[INFO] Installing package: qt5
[DEBUG] install(qt5, install_type=source, static=False)
[TRACE] Woohoo, this one's already cached (qt5)
[TRACE] Woohoo, this one's already cached (qt5)
[TRACE] qt5 has cached installed-status: False
[TRACE] Woohoo, this one's already cached (qt5)
[TRACE] Woohoo, this one's already cached (qt5)
[DEBUG] Using packager source
[WARNING] Cannot find a source URI for package qt5
[ERROR] Error installing package qt5. Aborting.

[log_qt5.log](https://github.com/gnuradio/pybombs/files/4038106/log_qt5.log)

anilgurses, 2020-01-15:
Same at Mac OS X Catalina. 
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package pyqt5:
Configuration failed
[ERROR] Error installing package pyqt5. Aborting.

roman-popenov, 2020-03-06:
I am seeing the same issue. 

I tried running:
`python3 configure.py --qmake /usr/local/Cellar/qt/5.14.1/bin/qmake` and then I saw the following error:

```bash
Querying qmake about your Qt installation...
Determining the details of your Qt installation...
This is the GPL version of PyQt 5.6 (licensed under the GNU General Public
License) for Python 3.7.6 on darwin.
Error: This version of PyQt5 and the commercial version of Qt have incompatible
licenses.
```
I was able to get around this by changing the line `  out << "Open Source" << '\\n';` to `  out << "Open Source" << '\\n';` in the **main** method.

The next error:

```bash
The Python dbus module doesn't seem to be installed.
Qt v5.14.1 (Open Source) is being used.
The qmake executable is /usr/local/Cellar/qt/5.14.1/bin/qmake.
Qt is built as a shared library.
SIP 5.1.1 is being used.
The sip executable is
/Library/Frameworks/Python.framework/Versions/3.7/bin/sip5.
These PyQt5 modules will be built: QtCore, QtGui, QtHelp, QtMultimedia,
QtMultimediaWidgets, QtNetwork, QtOpenGL, QtPrintSupport, QtQml, QtQuick,
QtSql, QtSvg, QtTest, QtWidgets, QtXml, QtXmlPatterns, QtDesigner, QtDBus,
_QOpenGLFunctions_2_0, _QOpenGLFunctions_2_1, _QOpenGLFunctions_4_1_Core,
QtSensors, QtSerialPort, QtBluetooth, QtMacExtras, QtPositioning,
QtQuickWidgets, QtWebSockets, QtWebChannel, QtWebEngineWidgets, QtLocation,
QtNfc, QtWebEngineCore.
The PyQt5 Python package will be installed in
/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages.
PyQt5 is being built with generated docstrings.
PyQt5 is being built with 'protected' redefined as 'public'.
The Designer plugin will be installed in
/usr/local/Cellar/qt/5.14.1/plugins/designer.
The qmlscene plugin will be installed in
/usr/local/Cellar/qt/5.14.1/plugins/PyQt5.
The PyQt5 PEP 484 stub files will be installed in
/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/PyQt5.
The PyQt5 .sip files will be installed in
/Library/Frameworks/Python.framework/Versions/3.7/share/sip/PyQt5.
pyuic5, pyrcc5 and pylupdate5 will be installed in
/Library/Frameworks/Python.framework/Versions/3.7/bin.
The interpreter used by pyuic5 is python3.7.
Generating the C++ source for the QtCore module...
Error: Unable to create the C++ code.
```

I have downloaded the Qt5.14.1 source code, I am able to install it from the binaries, but then I run into a similar issue with `PyYaml` module:
```bash
[WARNING] Configuration failed. Re-trying with higher verbosity.
/bin/sh: ./configure: No such file or directory
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package pyyaml:
Configuration failed
[ERROR] Error installing package pyyaml. Aborting.
```