gnuradio/pybombs #569: Recommended install via PyBombs fails on macOS High Sierra

devnulling, 2020-01-21:
Moving this issue to here from: https://github.com/gnuradio/gnuradio/issues/1531

# System versions
System version: macOS 10.13.1
Package manager: Homebrew (due to an OOT module needing PythonNet, macports can't be used)
Python version: Python 2.7.14

# How to recreate
After creating an installation prefix (step 4 of the quick-start):
```
$ pybombs install gnuradio
```

# Output
```
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
Password:
Collecting PyGTK
  Using cached pygtk-2.24.0.tar.bz2
    Complete output from command python setup.py egg_info:
    ********************************************************************
    * Building PyGTK using distutils is only supported on windows. *
    * To build PyGTK in a supported way, read the INSTALL file.    *
    ********************************************************************

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/tmp/pip-build-OyRkiY/PyGTK/
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
PyBOMBS.install_manager - ERROR - Package has no install method: libevent
```

# Attempted fix
```
$ pybombs install libevent
PyBOMBS - INFO - PyBOMBS Version 2.3.2
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
PyBOMBS.install_manager - ERROR - Package has no install method: libevent
```

```
$ brew install libevent
==> Downloading https://homebrew.bintray.com/bottles/libevent-2.1.8.high_sierra.bottle.tar.gz
######################################################################## 100.0%
==> Pouring libevent-2.1.8.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/libevent/2.1.8: 847 files, 2.2MB
```
```
$ brew install libevent
Warning: libevent 2.1.8 is already installed
```

Same error afterwards.

michaelld, 2020-01-21:
@nsbruce tagging the OP (or trying to ...)

michaelld, 2020-01-21:
I would be -very- surprised if PyBOMBS actually works on OSX, given that there are no internal references to `DYLD_LIBRARY_*` ... there are to the Unix-y `LD_LIBRARY_*`, which leads me to believe that there is library path manipulation & hence why on OSX I doubt it will work.

michaelld, 2020-01-21:
Is pythonnet this: https://pythonnet.github.io ...? If so, I'll see if I can add it to MacPorts.

nsbruce, 2020-01-21:
The tag worked @michaelld. I see what you mean about the references. It's been a while since I tried to do this. I'm not sure if it's still a relevant issue since that's two macOS versions old. I think I switched to a dockerized-GR when this didn't work.

Whether for this issue or not, having pythonnet available on MacPorts would be useful!