gnuradio/pybombs #570: Failed to install gnuradio-default(3.8) follow the "To run PyBOMBS (Quickstart)"

takurx, 2020-01-28:
I have installed follow "To run PyBOMBS (Quickstart)".
I saw error of ModuleNotFoundError.

[image]

It generate module of gnuradio under "~/{base_folder}/lib/python3/dist-package".
I added $PYTHONPATH in setup_env.sh this path that is "/home/{user_name}/{base_folder}/lib/python3/dist-package".
And GNU radio 3.8 started.

setup_env.sh is not enough $PYTHONPATH when "pybombs auto-config" generated. Thanks.

takurx, 2020-02-06:
#553 is same problem.

Phil67ago, 2022-12-04:
I fixed my error by updating pybombs with following patch in '~/.local/usr/lib/python3.10/site-packages/pybombs'.
[pybombs.patch.txt](https://github.com/gnuradio/pybombs/files/10148749/pybombs.patch.txt)
Then update 'setup_env.sh' by issue following in --- your working pybombs run directory ---:
pybombs prefix write-env

Secondly I could not get LD_LIBRARY_PATH working, I don't know why but it won't take that in when 'ld' loads things.
This is on Linux/Fedora 36.

Solved by:
echo "--- your working pybombs run directory ---/gnuradio-3.11.0/lib64/" > /etc/ld.so.conf.d/gnuradio-3.11.conf
Close everything and logout and in again.