gnuradio/pybombs #571: Installing pybombs starts to play pacman

TylerTucker, 2020-01-31:
When installing pyBOMBS on Ubuntu, something strange happens...the installer starts to play Pacman. The system I was installing on so happens to have the classic game installed, but I guess the installer tries executing this expecting this to be an Arch Linux system! The installer hangs as its trying to beat my high score - even closing it causes it to reopen again and again.

[image]

I used pyBOMBS 2.3.4a0 on Ubuntu 18.04 LTS for reference.

Architector4, 2020-03-26:
Yeah, the problem seems to be at https://github.com/gnuradio/pybombs/blob/master/pybombs/packagers/pacman.py#L37

Judging by the looks it just checks to see if a program called `pacman` can be run, and, if so, assumes Arch Linux's `pacman` is available and runs it with various commands.

mbr0wn, 2020-07-28:
This is the most hilarious bug I've ever seen.

jkbecker, 2020-10-03:
I'm very torn between fixing this and keeping this open forever because it's so amazing. Who doesn't like a surprise game of pacman... :D