gnuradio/pybombs #572: Installing gnuradio fails on building pyyaml

cafeclimber, 2020-03-06:
Before posting an issue, please check the following:

- [x] Is this actually a bug in PyBOMBS, or is it a bug in the package you're
      installing? If it's the latter, this is the wrong place to post. It won't
      get fixed unless the author of the package is informed.

- [x] How can this issue be reproduced? We have Docker-based tests for various
      Linux distributions, which are the reference for whether or not we
      consider a specific distribution as functional. Have you provided
      instructions to reproduce?

- [x] Have you included all relevant logs to help developers find the issue?
      Did you bump the verbosity (`-v` or `-vv`)?

Error description:
Using pybombs to install gnuradio fails when it tries to configure pyyaml.

System info:
OS: Arch Linux
Kernel: 5.5.7-arch1-1
PyBOMBS: 2.3.4a0
Python: 3.8.1

Commands used:
```zsh
$ sudo -H pip3 install --upgrade git+http://github.com/gnuradio/pybombs.git
$ pybombs prefix init ./grmaint38_uhd315 -a grmaint38_uhd315
$ pybombs -v -p grmaint38_uhd315 recipes add-defaults
$ pybombs -vv -p grmaint38_uhd315 install gnuradio
```

Output:
```
Install tree:
|
\- gnuradio
   |
   +- uhd
   |  |
   |  \- python-requests
   |
   \- pyyaml
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[TRACE] pyyaml has cached installed-status: False
[INFO] Installing package: pyyaml
[DEBUG] install(pyyaml, install_type=source, static=False)
[TRACE] Woohoo, this one's already cached (pyyaml)
[TRACE] Woohoo, this one's already cached (pyyaml)
[TRACE] pyyaml has cached installed-status: False
[TRACE] Woohoo, this one's already cached (pyyaml)
[TRACE] Woohoo, this one's already cached (pyyaml)
[DEBUG] Using packager source
[DEBUG] State on package pyyaml is 10
[DEBUG] Package pyyaml is already fetched.
[DEBUG] Requiring packages on host system: []
[DEBUG] Attempting to add binary package manager apt
[DEBUG] Attempting to add binary package manager yumdnf
[DEBUG] Attempting to add binary package manager port
[DEBUG] Attempting to add binary package manager brew
[DEBUG] Attempting to add binary package manager zypper
[DEBUG] Attempting to add binary package manager pacman
[DEBUG] pacman is supported!
[DEBUG] Attempting to add binary package manager portage
[DEBUG] Attempting to add binary package manager pymod
[DEBUG] pymod is supported!
[DEBUG] Attempting to add binary package manager pip
[DEBUG] Using pip executable: pip3
[DEBUG] pip is supported!
[DEBUG] Attempting to add binary package manager pkgconfig
[DEBUG] pkgconfig is supported!
[DEBUG] Attempting to add binary package manager cmd
[DEBUG] cmd is supported!
[DEBUG] Using binary packagers: ['pacman', 'pymod', 'pip', 'pkgconfig', 'cmd']
[DEBUG] Phase 1: Creating install tree and installing binary packages:
[TRACE] Checking if package `build-essential' goes into tree...
[DEBUG] Checking if package build-essential is installable...
[TRACE] Loading recipe file: /home/bayleef/.virtualenvs/sdr-getting-started/lib/python3.8/site-packages/pybombs/recipes/build-essential.lwr
[TRACE] Loading template file: /home/bayleef/.virtualenvs/sdr-getting-started/lib/python3.8/site-packages/pybombs/templates/empty.lwt
[TRACE] Inheriting from file /home/bayleef/.virtualenvs/sdr-getting-started/lib/python3.8/site-packages/pybombs/templates/empty.lwt
[TRACE] Loaded recipe - build-essential
[TRACE] Woohoo, this one's already cached (build-essential)
[TRACE] Woohoo, this one's already cached (build-essential)
[TRACE] exists(build-essential)
[TRACE] Parsing requirements string: None
[TRACE] exists(build-essential)
[TRACE] Parsing requirements string: None
[TRACE] exists(build-essential)
[TRACE] Parsing requirements string: None
[TRACE] exists(build-essential)
[TRACE] Parsing requirements string: None
[TRACE] exists(build-essential)
[TRACE] Parsing requirements string: make --version && (gcc --version || clang --version) && (g++ --version || clang --version)
[TRACE] Calling ev for recursive satisfier rule evaluation
[DEBUG] Found version number: 4.3
[DEBUG] Found version number: 9.2.1
[DEBUG] Found version number: 9.2.1
[DEBUG] Checking if package build-essential is installed...
[TRACE] Woohoo, this one's already cached (build-essential)
[TRACE] Woohoo, this one's already cached (build-essential)
[TRACE] Woohoo, this one's already cached (build-essential)
[TRACE] Checking if recipe build-essential is installed
[TRACE] Parsing requirements string: None
[TRACE] Checking if recipe build-essential is installed
[TRACE] Parsing requirements string: None
[TRACE] Checking if recipe build-essential is installed
[TRACE] Parsing requirements string: None
[TRACE] Checking if recipe build-essential is installed
[TRACE] Parsing requirements string: None
[TRACE] Checking if recipe build-essential is installed
[TRACE] Parsing requirements string: make --version && (gcc --version || clang --version) && (g++ --version || clang --version)
[TRACE] Calling ev for recursive satisfier rule evaluation
[DEBUG] Found version number: 4.3
[DEBUG] Found version number: 9.2.1
[DEBUG] Found version number: 9.2.1
[TRACE] Installed, but no update requested. Does not go into tree.
[DEBUG] Phase 1 complete: All binary dependencies installed.
[DEBUG] Phase 2: Recursively installing source packages to prefix:
[DEBUG] Phase 2 complete: All source packages installed.
[DEBUG] Requirements met.
[DEBUG] Using build directory: /home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315/src/pyyaml
[DEBUG] Configuring recipe pyyaml
[DEBUG] Using vars - ordereddict([('config_opt', ''), ('builddir', '/home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315/src/pyyaml')])
[DEBUG] In cwd - /home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315/src/pyyaml
[DEBUG] Executing command `$ ./configure --prefix=/home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315'
/bin/sh: ./configure: No such file or directory
[DEBUG] Thread signaled termination or returned
[DEBUG] Return value: 127
[WARNING] Configuration failed. Re-trying with higher verbosity.
[DEBUG] Configuring recipe pyyaml
[DEBUG] Using vars - ordereddict([('config_opt', ''), ('builddir', '/home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315/src/pyyaml')])
[DEBUG] In cwd - /home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315/src/pyyaml
[DEBUG] Executing command `$ ./configure --prefix=/home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315'
/bin/sh: ./configure: No such file or directory
[DEBUG] Thread signaled termination or returned
[DEBUG] Return value: 127
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package pyyaml:
Configuration failed
[ERROR] Error installing package pyyaml. Aborting.
```

jai2410, 2020-03-10:
I am having the same issue.
 
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 3.8.0
PyBOMBS.prefix - INFO - Installing default packages for prefix...
PyBOMBS.prefix - INFO -
  - gnuradio
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
   |
   +- pyyaml
   |
   \- pyqt5
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: pyqt5
Configuring: (100%) [=========================================================]
PyBOMBS.Packager.source - WARNING - Configuration failed. Re-trying with higher verbosity.
Error: Use the --qmake argument to explicitly specify a working Qt qmake.
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package pyqt5:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package pyqt5. Aborting.

velichkov, 2020-03-10:
Hi @cafeclimber, @jai2410,

For Arch Linux the fixes for `pyyaml` and `pyqt5` are probably similar to the fix for `python-click-plugins` (https://github.com/gnuradio/gr-recipes/pull/178).

Open `~/.pybombs/recipes/gr-recipes/pyyaml.lwr` and apply the following change.
```diff
diff --git a/pyyaml.lwr b/pyyaml.lwr
index e3c408b..ccc34fd 100644
--- a/pyyaml.lwr
+++ b/pyyaml.lwr
@@ -18,15 +18,19 @@
 #
 
 category: baseline
-inherit: autoconf
+inherit: distutils
 depends:
 - python
 satisfy:
   deb: python-yaml
   rpm: python2-pyyaml
+  packman: python2-pyaml
+  pip: PyYAML
 satisfy@python3:
   deb: python3-pyyaml || python3-yaml
   rpm: python3-pyyaml
+  packman: python-pyaml
+  pip: PyYAML
 description: Python wrapper for YAML parser
 source: http://pyyaml.org/download/pyyaml/PyYAML-3.12.tar.gz
```
changing `inherit: autoconf` to `inherit: distutils` should fix the `/bin/sh: ./configure: No such file or directory` error when building from source.

Open `~/.pybombs/recipes/gr-recipes/pyqt5.lwr` and apply the following change.
```diff
diff --git a/pyqt5.lwr b/pyqt5.lwr
index 3842da2..3941da9 100644
--- a/pyqt5.lwr
+++ b/pyqt5.lwr
@@ -28,10 +28,13 @@ install_like: qt5
 satisfy:
   deb: ( pyqt5-dev >= 5.3.2 ) && ( pyqt5-dev-tools >= 5.3.2 ) && python-pyqt5
   rpm: (python-qt5 || python2-qt5) && python-qt5-devel
-  pacman: python-pyqt5
+  pacman: python2-pyqt5
   port: py27-qt5 >= 5.7.1
   portage: dev-python/PyQt5 >= 5.7.1
+  pip: PyQt5
 satisfy@python3:
   deb: python3-pyqt5
   rpm: python3-qt5 && python3-qt5-devel
+  pacman: python-pyqt5
+  pip: PyQt5
 source: wget+https://sourceforge.net/projects/pyqt/files/PyQt5/PyQt-5.6/PyQt5_gpl-5.6.tar.gz
```

jai2410, 2020-03-11:
@velichkov I missed one information to mention:I am using Mac OS:Catalina 10.15.12.

michaelld, 2020-03-11:
@jai2410 I'd be very surprised if PyBOMBS works on OSX. It requires setting the dynamic library path search, which on Linux uses LD_LIBRARY_PATH ... on OSX this is DYLD_LIBRARY_PATH ... which is nowhere in the codebase; like, at all. My guess is the tweaks to get PyBOMBS working on OSX aren't that significant ... it's more a matter of having time to do them!

All of this said: if it works for you ... well ...great!

cafeclimber, 2020-03-11:
Ugh sorry to be a pain, but maybe after going through all this, I will have submitted all the PRs to get Arch support working. That fix worked for pyyaml, but here's the newest issue after following the steps detailed above:
```
Install tree:
|
\- gnuradio
   |
   \- uhd
      |
      \- python-requests
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[TRACE] python-requests has cached installed-status: False
[INFO] Installing package: python-requests
[DEBUG] install(python-requests, install_type=source, static=False)
[TRACE] Woohoo, this one's already cached (python-requests)
[TRACE] Woohoo, this one's already cached (python-requests)
[TRACE] python-requests has cached installed-status: False
[TRACE] Woohoo, this one's already cached (python-requests)
[TRACE] Woohoo, this one's already cached (python-requests)
[DEBUG] Using packager source
[DEBUG] State on package python-requests is 0
[DEBUG] Fetching source for recipe: python-requests
[DEBUG] Switching to src directory: /home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315/src
[TRACE] Trying to fetch wget+https://pypi.python.org/packages/source/r/requests/requests-2.8.1.tar.gz
[DEBUG] Getting fetcher for https://pypi.python.org/packages/source/r/requests/requests-2.8.1.tar.gz, type wget
[DEBUG] Using fetcher <pybombs.fetchers.wget.Wget object at 0x7f0c18ed1e20>
[DEBUG] Requiring packages on host system: ['python-requests']
[DEBUG] Attempting to add binary package manager apt
[DEBUG] Attempting to add binary package manager yumdnf
[DEBUG] Attempting to add binary package manager port
[DEBUG] Attempting to add binary package manager brew
[DEBUG] Attempting to add binary package manager zypper
[DEBUG] Attempting to add binary package manager pacman
[DEBUG] pacman is supported!
[DEBUG] Attempting to add binary package manager portage
[DEBUG] Attempting to add binary package manager pymod
[DEBUG] pymod is supported!
[DEBUG] Attempting to add binary package manager pip
[DEBUG] Using pip executable: pip3
[DEBUG] pip is supported!
[DEBUG] Attempting to add binary package manager pkgconfig
[DEBUG] pkgconfig is supported!
[DEBUG] Attempting to add binary package manager cmd
[DEBUG] cmd is supported!
[DEBUG] Using binary packagers: ['pacman', 'pymod', 'pip', 'pkgconfig', 'cmd']
[DEBUG] Phase 1: Creating install tree and installing binary packages:
[TRACE] Checking if package `python-requests' goes into tree...
[TRACE] python-requests has cached installable-status: True
[TRACE] python-requests has cached installed-status: False
[DEBUG] Testing binary install for package python-requests.
[DEBUG] install(python-requests, install_type=binary, static=False)
[TRACE] Woohoo, this one's already cached (python-requests)
[TRACE] Woohoo, this one's already cached (python-requests)
[TRACE] Woohoo, this one's already cached (python-requests)
[TRACE] Woohoo, this one's already cached (python-requests)
[DEBUG] Using packager pacman
[TRACE] install(python-requests, static=False)
[TRACE] Parsing requirements string: None
[DEBUG] Using packager pymod
[TRACE] install(python-requests, static=False)
[TRACE] Parsing requirements string: None
[DEBUG] Using packager pip
[TRACE] install(python-requests, static=False)
[TRACE] Parsing requirements string: requests
[TRACE] Calling ev for recursive satisfier rule evaluation
[DEBUG] Calling `pip3 install requests'
[DEBUG] Running with elevated privileges.
[INFO] Executing command with elevated privileges: `sudo -H pip3 install requests'
Requirement already satisfied: requests in /home/bayleef/.virtualenvs/sdr-getting-started/lib/python3.8/site-packages (2.23.0)
Requirement already satisfied: idna<3,>=2.5 in /home/bayleef/.virtualenvs/sdr-getting-started/lib/python3.8/site-packages (from requests) (2.9)
[... 113 lines cut ...]
[DEBUG] Executing command `$ python setup.py install --prefix=/home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315'
Warning: 'classifiers' should be a list, got type 'tuple'
running install
error: can't create or remove files in install directory

The following error occurred while trying to add or remove files in the
installation directory:

    [Errno 2] No such file or directory: '/home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315/lib/python3.8/site-packages/test-easy-install-8245.write-test'

The installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was:

    /home/bayleef/Documents/Programming/gnuradio/grmaint38_uhd315/lib/python3.8/site-packages/

This directory does not currently exist.  Please create it and try again, or
choose a different installation directory (using the -d or --install-dir
option).

[DEBUG] Thread signaled termination or returned
[DEBUG] Return value: 1
[ERROR] Problem occurred while building package python-requests:
Installation failed
[ERROR] Error installing package python-requests. Aborting.
```

velichkov, 2020-03-11:
Hi @cafeclimber,

> Ugh sorry to be a pain, 

No problem, don't worry. 

>That fix worked for pyyaml, but here's the newest issue after following the steps detailed above:
> 
> ```
> [ERROR] Problem occurred while building package python-requests:
> ```

The fix for `python-request` is also similar but its recipe is in a different directory. You can find where the `python-request.lwr` is using the following command
```
pybombs recipes list -l python-requests
```
then you need open it and add the following
```diff
diff --git a/pybombs/recipes/python-requests.lwr b/pybombs/recipes/python-requests.lwr
index b609d8e..b3664bf 100644
--- a/pybombs/recipes/python-requests.lwr
+++ b/pybombs/recipes/python-requests.lwr
@@ -24,10 +24,14 @@ satisfy:
   deb: python-requests
   rpm: python-requests || python2-requests
   port: py27-requests
+  pacman: python2-requests
+  pip: requests
 # This is an override for Python 3. It will supersede the regular `satisfy`, but
 # only for the keys listed here (and of course only when the target is using
 # Python 3).
 satisfy@python3:
   deb: python3-requests
   rpm: python3-requests
+  pacman: python-requests
+  pip: requests
 source: wget+https://pypi.python.org/packages/source/r/requests/requests-2.8.1.tar.gz
```

cafeclimber, 2020-03-11:
Awesome! That worked. Thank you. It also threw an error generating build files because I didn't have mako installed. I'll try to submit a PR to include that as a build requirement in addition to my PR for the above fixes. Now to hope it builds :)

closed 2020-03-11 by cafeclimber