gnuradio/pybombs #573: GNURadio 3.8 inPyBombs fails to build on Ubuntu 20.04

cyrus104, 2020-04-09:
Good day,

I understand that Ubuntu 20.04 is still in beta but we are starting to move several project over to the new LTS, we expect a lot of testing will be needed and need to start working through the issues with the release being so close.

We are running into build issues when install pybombs and initializing it. It looks like the issues start with pygtk and then continue from there.

Install command:
sudo pip3 install --upgrade git+https://github.com/gnuradio/pybombs.git
pybombs auto-config
pybombs recipes add-defaults
pybombs prefix init ~/sdr -R gnuradio-default -a master

[INFO] Executing command with elevated privileges: sudo -H pip3 install PyGTK'
 Collecting PyGTK
 Downloading pygtk-2.24.0.tar.bz2 (2.4 MB)
 ERROR: Command errored out with exit status 1:
 command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-7wfymupz/PyGTK/setup.py'"'"'; __file__='"'"'/tmp/pip-install-7wfymupz/PyGTK/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-7wfymupz/PyGTK/pip-egg-info
 cwd: /tmp/pip-install-7wfymupz/PyGTK/
Complete output (4 lines):

Building PyGTK using distutils is only supported on windows.
To build PyGTK in a supported way, read the INSTALL file.  

ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

`checking for PYGOBJECT... no
 configure: error: Package requirements (pygobject-2.0 >= 2.21.3) were not met:
 No package 'pygobject-2.0' found
 Consider adjusting the PKG_CONFIG_PATH environment variable if you
 installed software in a non-standard prefix.
 Alternatively, you may set the environment variables PYGOBJECT_CFLAGS
 and PYGOBJECT_LIBS to avoid the need to call pkg-config.
 See the pkg-config man page for more details.
 [ERROR] Configuration failed after running at least twice.
 [ERROR] Problem occurred while building package pygtk:
 Configuration failed
 [ERROR] Error installing package pygtk. Aborting.`

argilo, 2020-04-28:
I got the same error when trying to build on Ubuntu 20.04, and was able to get past it by deleting the `- pygtk` line from `~/.pybombs/recipes/gr-recipes/gnuradio.lwr`. I also had to change the `PYTHONPATH` line in `prefix/setup_env.sh` to the following:

`export PYTHONPATH="/home/argilo/prefix_38/lib/python3.8/site-packages:/home/argilo/prefix_38/lib/python3.8/dist-packages:/home/argilo/prefix_38/lib/python3/site-packages:/home/argilo/prefix_38/lib/python3/dist-packages:$PYTHONPATH"`

argilo, 2020-04-28:
Also I noticed that the gqrx recipe attempts to build qt4 from source; removing the `- qt4` line from `~/.pybombs/recipes/gr-recipes/gqrx.lwr` fixed that.

rowb1, 2020-07-03:
Worked for me.
Linux Ubuntu-20 5.4.0-40-generic #44-Ubuntu SMP Tue Jun 23 00:01:04 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

**nano ~/.pybombs/recipes/gr-recipes/gnuradio.lwr**

remove old files from faulty builds before remaking

pybombs prefix init ~/gr38 -R gnuradio-default

depends:
...
- qt5
- qwt6
- numpy
- lxml
# - pygtk
- pycairo
- pyqt5
...

qwrd, 2020-07-22:
@argilo I tried to build this recently on Kali rolling, and hit that same bug about qt4.
Manually removing qt4 dependency from gqrx.lrw fixed the issue.

marxenegls, 2020-08-23:
it worked , the setup_env.sh , ran issues with the build after 10hrs on VBox 3Gbram 2cpu - and yes i did fork them to use my creds - installed. deps of cppunit and gnuradio-runtime  and i'll give the build another run

argilo, 2020-10-05:
Closing, as this was fixed in https://github.com/gnuradio/gr-recipes/pull/186.

closed 2020-10-05 by argilo