gnuradio/pybombs #574: PyBOMBS fails to detect scikit-commpy via pip

jdemel, 2020-05-07:
On Ubuntu 20.04. Just run `pybombs -vvv install scikit-commpy`.
Expected: Install scikit-commpy via pip.
Output: Package not found. pybombs tries to install scikit-commpy from source.

I updated the recipe as well. But that didn't help either.
```
category: baseline
inherit: distutils
depends:
- scipy
- numpy
satisfy:
    pip: scikit-commpy >= 0.3
```
pip3 reports a successfully installed scikit-commpy but PyBOMBS doesn't seem to care.
```
<snap>
[DEBUG] Calling `pip3 install scikit-commpy'
[DEBUG] Running with elevated privileges.
[INFO] Executing command with elevated privileges: `sudo -H pip3 install scikit-commpy'
Requirement already satisfied: scikit-commpy in /usr/local/lib/python3.8/dist-packages (0.5.0)
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (from scikit-commpy) (1.17.4)
Requirement already satisfied: matplotlib in /usr/lib/python3/dist-packages (from scikit-commpy) (3.1.2)
Requirement already satisfied: scipy in /usr/lib/python3/dist-packages (from scikit-commpy) (1.3.3)
[DEBUG] Thread signaled termination or returned
[DEBUG] Return value: 0
<snap>
```
I tried to find out why pybombs fails to detect it but I can't find the reason.

pauly617, 2020-07-27:
I believe this is why its not working, python-scipy has been changed to python3-scipy, also python3-scapy and python3-numpy.

So i just came here to find out, where do I make the build changes in the build file? I am not familiar with this Program pybombs but I assume changing the build to look for the correct names will solve this. This is also just a guess, but does anyone know off hand how to alter what ever part of the program looks for these dependency?

Thank You~

dkozel, 2020-07-27:
Hi @pauly617
You need to modify the recipe file(s). I saw your post on the Issue where we are looking for a new maintainer. Part of the handling that needs to improve is Python 2 vs 3. PyBOMBs is still targeting GNU Radio versions, and operating systems, which have or require Python 3.
https://github.com/gnuradio/gr-recipes

jdemel, 2021-01-22:
I still see this error. I just checked on my Ubuntu 20.04 install
```bash
$ pip3 show scikit-commpy
Name: scikit-commpy
Version: 0.6.0
Summary: Digital Communication Algorithms with Python
Home-page: http://veeresht.github.com/CommPy
Author: None
Author-email: None
License: BSD 3-Clause
Location: /usr/local/lib/python3.8/dist-packages
Requires: numpy, scipy, matplotlib
Required-by:
```
But if I do
```
pybombs -vvv install scikit-commpy
```
The results are still wrong.
```
[DEBUG] Reading config info from file: /etc/pybombs/config.yml
[DEBUG] Parsing config file failed (/etc/pybombs/config.yml).
[DEBUG] Reading config info from file: ~/.pybombs/config.yml
...
[INFO] Prefix Python version is: 3.8.5
[INFO] PyBOMBS Version 2.3.4
...
[DEBUG] Attempting to add binary package manager apt
[DEBUG] apt is supported!
[DEBUG] Attempting to add binary package manager pymod
[DEBUG] pymod is supported!
[DEBUG] Attempting to add binary package manager pip
[DEBUG] Using pip executable: pip3
[DEBUG] pip is supported!
[DEBUG] Attempting to add binary package manager pkgconfig
[DEBUG] pkgconfig is supported!
[DEBUG] Attempting to add binary package manager cmd
[DEBUG] cmd is supported!
[DEBUG] Using binary packagers: ['apt', 'pymod', 'pip', 'pkgconfig', 'cmd']
[INFO] Phase 1: Creating install tree and installing binary packages:
[TRACE] Checking if package `scikit-commpy' goes into tree...
[DEBUG] Checking if package scikit-commpy is installable...
[TRACE] Loading recipe file: ~/.pybombs/recipes/gr-etcetera/scikit-commpy.lwr
[TRACE] Loading template file: /usr/local/lib/python3.8/dist-packages/pybombs/templates/distutils.lwt
[TRACE] Inheriting from file /usr/local/lib/python3.8/dist-packages/pybombs/templates/distutils.lwt
[TRACE] Loading template file: /usr/local/lib/python3.8/dist-packages/pybombs/templates/empty.lwt
[TRACE] Inheriting from file /usr/local/lib/python3.8/dist-packages/pybombs/templates/empty.lwt
[TRACE] Loaded recipe - scikit-commpy
[TRACE] Woohoo, this one's already cached (scikit-commpy)
[TRACE] Woohoo, this one's already cached (scikit-commpy)
[DEBUG] Checking if package scikit-commpy is installed...
[TRACE] Woohoo, this one's already cached (scikit-commpy)
[TRACE] Woohoo, this one's already cached (scikit-commpy)
[TRACE] Woohoo, this one's already cached (scikit-commpy)
[TRACE] Checking if recipe scikit-commpy is installed
[TRACE] Parsing requirements string: None
[TRACE] Checking if recipe scikit-commpy is installed
[TRACE] Parsing requirements string: None
[TRACE] Checking if recipe scikit-commpy is installed
[TRACE] Parsing requirements string: scikit-commpy >= 0.3
[TRACE] Calling ev for recursive satisfier rule evaluation
[DEBUG] Loading pip install cache.
[TRACE] Checking if recipe scikit-commpy is installed
[TRACE] Parsing requirements string: None
[TRACE] Checking if recipe scikit-commpy is installed
[TRACE] Parsing requirements string: None
[DEBUG] Package scikit-commpy is not installed.
[DEBUG] Testing binary install for package scikit-commpy.
[DEBUG] install(scikit-commpy, install_type=binary, static=False)
[TRACE] Woohoo, this one's already cached (scikit-commpy)
[... 88 lines cut ...]
\- scikit-commpy
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[TRACE] scikit-commpy has cached installed-status: False
[INFO] Installing package: scikit-commpy
[DEBUG] install(scikit-commpy, install_type=source, static=False)
[TRACE] Woohoo, this one's already cached (scikit-commpy)
[TRACE] Woohoo, this one's already cached (scikit-commpy)
[TRACE] scikit-commpy has cached installed-status: False
[TRACE] Woohoo, this one's already cached (scikit-commpy)
[TRACE] Woohoo, this one's already cached (scikit-commpy)
[DEBUG] Using packager source
[DEBUG] State on package scikit-commpy is 0
[DEBUG] Fetching source for recipe: scikit-commpy
[DEBUG] Switching to src directory: [prefix folder]
[TRACE] Trying to fetch wget -r https://pypi.org/project/scikit-commpy/#files
[DEBUG] That didn't work.
[DEBUG] Unrecognized URI: wget -r https://pypi.org/project/scikit-commpy/#files
[ERROR] Problem occurred while building package scikit-commpy:
Unable to fetch recipe scikit-commpy
[ERROR] Error installing package scikit-commpy. Aborting.
```
I'm happy it fails to install from source because 1. it is already installed. 2. pybombs missed its detection.

Is there a bug in pybombs that prevents pybombs from interacting with pip correctly?

pauly617, 2021-02-01:
I just tried installing it and I dont see this error. How ever I didn t try on a Ubuntu Box maybe thats part of the issue. 

I tried on Kali 5.9.15. 
I have not upgraded to kali 5.10 because wireless will not work for me and I havent had time to figure  out why.