gnuradio/pybombs #577: Find new maintainer for PyBOMBS

mbr0wn, 2020-06-04:
Interested?

pauly617, 2020-07-27:
Is it a difficult thing to maintain this? I have the time and would be glad to if you can explain what ever is needed ill be able to absorb it. I dont have allot of programing knowledge but Ive been a Net Admin since the time when they still called it Net Admin LOL... Plus I been working on the command line since 1977-78..  It might even be something I already know how to do its just not the path of Computers I mastered in..  Even if its just temporary for you to find another Ill give it a shot so hit me up here or email me ptla617@gmail

Also I wanted to add the new build for gr-gsm on kali boxes and maybe regular Debian based systems python-scapy python-scipy and python-numpy all have been name changed and I believe altering the build script or build file and updating it to python3-scapy python3-scipy etc... I know for a fact python-scipy is one of the dependency's and the other Two I believe are also even if not directly needed for pybombs to build gr-gsm they are needed to run gr-gsm in the configuration needed for gnuradio to do GSM stuff with rtl-sdr's and HackRF-One etc...

argilo, 2020-10-05:
I'll close this off, since I and @jkbecker have agreed to maintain PyBOMBS.

closed 2020-10-05 by argilo

pauly617, 2020-10-05:
OK guys I just you know I want to try to help but I still will if you need
it I’m learning a lot about telephone stack systems all open source of
course maybe a few proprietary pieces in there but it’s all me in the open
source so if you need some help just yeah I’m really good at fixing things
I’m not a creator of an engineer that takes creative things and makes new
things out of them good luck guys