gnuradio/pybombs #579: Pybombs fails to install GNURadio 3.8 on Ubuntu 18.04

danalvarez, 2020-06-28:
Hello everyone:

Posting this here because I cannot seem to find any solution online, been trying this on and off for several weeks.

So I have a working installation of GNURadio 3.7 on the `default/` prefix. I did this with Pybombs and no issues after working out some quirks. I want to give 3.8 a try so I've been trying to install this into a new prefix, but it just doesn't work. I feel like the error is kind of obscure, I don't know what it relates to. 

Always fails after `[ 39%] Building CXX object gr-blocks/swig/CMakeFiles/blocks_swig0.dir/CMakeFiles/blocks_swig0.dir/blocks_swig0PYTHON_wrap.cxx.o`. I'm attaching a full log of the output...

Running `pybombs inv` after the failed installation shows:
```
[INFO] Prefix Python version is: 3.6.9
[INFO] PyBOMBS Version 2.3.4a0
Showing package state:
uhd:                installed
gnuradio:           configured
```

Running `pybombs inv` in the `default/` prefix shows:

```
[INFO] Prefix Python version is: 3.6.9
[INFO] PyBOMBS Version 2.3.4a0
Showing package state:
uhd:                installed
gnuradio:           installed
pyconstruct:        installed
libfec:             installed
gr-satellites37:    installed
gr-gpredict-doppler37:installed
rtl-sdr:            installed
hackrf:             installed
osmo-sdr:           installed
gr-osmosdr37:       installed
gr-msgtools:        installed
gr-sattools:        installed
```

## Steps to reproduce

1. Install GNURadio 3.7 into `default/` prefix
2. `pybombs prefix init ~/prefix/gr38 -R gnuradio-default`

I appreciate anyone's input on this... I feel stumped and don't know what else to search for.

Best wishes,

Dan

dkozel, 2020-06-28:
> [WARNING] A source build for package gnuradio was requested, but binary install was found!
Install gnuradio from source despite binary install available Y/[N]? y

Do you know where this binary version is? If you have a system wide version of GNU Radio installed 3.8 cannot be built into a prefix due to a bug as far as I can remember.

danalvarez, 2020-06-28:
Hey, @dkozel !

Thanks for the very quick response... You're completely right. I had installed GQRX via apt-get and this installs GNURadio system wide. I've removed it and now Pybombs correctly built 3.8 into the desired prefix.

Thanks!!!

closed 2020-06-28 by danalvarez