gnuradio/pybombs #582: Installing Gnuradio 3.8 via Pybombs

oayan, 2020-09-17:
I have followed the instructions on: https://github.com/gnuradio/pybombs. However, gnuradio companion is not starting.
This is the error I am getting
`/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
Traceback (most recent call last):
  File "/home/myUser/gr38/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/home/myUser/gr38/bin/gnuradio-companion", line 90, in run_main
    from gnuradio.grc.main import main
ModuleNotFoundError: No module named 'gnuradio.grc'
`

Has someone overcome this issue?

jkbecker, 2020-09-28:
Hi @oayan,

Could you post the output of the following commands please:

```
which python
python --version
which python3
python3 --version
pybombs config default_prefix
```

That will help with the troubleshooting. I haven't encountered this exact error but it smells like a python version issue to me...

oayan, 2020-09-29:
Hi @jkbecker, 
which python --> /usr/bin/python
python --version --> Python 2.7.17
which python3 --> /usr/bin/python3
python3 --version --> Python 3.6.9

We did not install using pybombs, but compiled GNUradio from source.

oayan, 2020-09-29:
which python --> /home/myUser/anaconda3/bin/python
which python3 --> which python3 /home/myUser/anaconda3/bin/python3
python --version --> Python 3.7.6
python3 --version --> Python 3.7.6
pybombs config default_prefix --> 
[INFO] Prefix Python version is: 3.6.9
[INFO] PyBOMBS Version 2.3.4
[INFO] Using config file: /home/myUser/.pybombs/config.yml
default_prefix: /home/myUser/gnuradio
  - Default Prefix

oayan, 2020-09-29:
I did not store the state at the date of this post, so I installed GNUradio from source afterwards. However, the instructions on the pybombs page above were  tried on a fresh Ubuntu 18.04 installation and it did not work out of the box.