gnuradio/pybombs #583: [Ubuntu 18.04/3.7] Repository http://git.alsa-project.org/alsa-lib.git/ not found

johnthagen, 2020-09-17:
On a clean Ubuntu 18.04.5 virtual machine, `alsa` fails to install. Tested at commit https://github.com/gnuradio/pybombs/commit/c437808e782357de16e3c71dd682959ac3ad3f5f

```bash
$ sudo apt install python python-pip python-virtualenv libcppunit-dev python-cheetah libqt4-dev libqwt5-qt4 libqwt-dev
$ sudo pip install --upgrade git+https://github.com/gnuradio/pybombs.git
$ pybombs auto-config
$ pybombs recipes add-defaults
$ pybombs prefix init ~/pybombs -R gnuradio-stable
...
[INFO] Executing command with elevated privileges: `sudo -H apt-get -y install libusb-1.0-0-dev'
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libdumbnet1
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  libusb-1.0-doc
The following NEW packages will be installed:
  libusb-1.0-0-dev libusb-1.0-doc
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 230 kB of archives.
After this operation, 1,774 kB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 libusb-1.0-0-dev amd64 2:1.0.21-2 [60.5 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 libusb-1.0-doc all 2:1.0.21-2 [170 kB]
Fetched 230 kB in 0s (527 kB/s)          
Selecting previously unselected package libusb-1.0-0-dev:amd64.
(Reading database ... 211300 files and directories currently installed.)
Preparing to unpack .../libusb-1.0-0-dev_2%3a1.0.21-2_amd64.deb ...
Unpacking libusb-1.0-0-dev:amd64 (2:1.0.21-2) ...
Selecting previously unselected package libusb-1.0-doc.
Preparing to unpack .../libusb-1.0-doc_2%3a1.0.21-2_all.deb ...
Unpacking libusb-1.0-doc (2:1.0.21-2) ...
Setting up libusb-1.0-doc (2:1.0.21-2) ...
Setting up libusb-1.0-0-dev:amd64 (2:1.0.21-2) ...
Install tree:
|
\- gnuradio
   |
   +- pygobject
   |
   +- uhd
   |
   +- libvolk
   |
   \- alsa
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: alsa
[INFO] Install python-apt to speed up apt processing.
fatal: repository 'http://git.alsa-project.org/alsa-lib.git/' not found
error: Could not fetch alsa
[ERROR] Unexpected error while fetching git+http://git.alsa-project.org/alsa-lib.git.
[ERROR] Command '['git', 'remote', 'update', 'alsa']' returned non-zero exit status 1
[ERROR] Problem occurred while building package alsa:
Unable to fetch recipe alsa
[ERROR] Error installing package alsa. Aborting.
```

argilo, 2020-10-05:
I'm confused why PyBOMBS would decide to install alsa from source. Could it be because you had not yet run `sudo apt update` to fetch the list of available packages?

johnthagen, 2020-10-05:
If I recall, this was run from a fully up to date Ubuntu 18.04 virtual machine, where I had run `apt update` and `apt upgrade` to ensure the host was fully up to date before starting any PyBombs steps.

argilo, 2020-10-05:
Strange. I tested this in an Ubuntu 18.04 container and it didn't attempt to build alsa or pygobject from source. We'll need to find out why that happened.

The also source URL is also broken though, so we should fix the recipe.