gnuradio/pybombs #589: Installation error

nbertamoni, 2020-10-16:
Hello, I don't really know from where the error is coming from, but I have had this issue trying to follow the read me to install gnuradio with pybombs.
Anyway...I've just installed Ubuntu 18.04 so i don have any problems with forgotten environment.
After finally installing all the drivers and updates to make peripherals work properly here Im

After the command: 
pybombs prefix init ~/gr37 -R gnuradio-stable
 I receive:

-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = TRUE
--   Dependency CHEETAH_FOUND = FALSE
--   Dependency LXML_FOUND = TRUE
--   Dependency PYGTK_FOUND = FALSE
--   Dependency NUMPY_FOUND = TRUE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:45 (GR_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/.../gr37/src/gnuradio/build/CMakeFiles/CMakeOutput.log".
See also "/home/.../gr37/src/gnuradio/build/CMakeFiles/CMakeError.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package gnuradio:
Configuration failed
[ERROR] Error installing package gnuradio. Aborting.

jkbecker, 2020-10-16:
Chi @nbertamoni, could you upload the `CMakeOutput.log` and `CMakeError.log` mentioned in your log to https://gist.github.com/ and paste the link here? Thanks!

nbertamoni, 2020-10-17:
Hello @jkbecker,
is that what you asked, right? if not, just tell me, please.

https://gist.github.com/nbertamoni/9f84cb985f33cd03c839accd00c29602