gnuradio/pybombs #598: Installing pybombs in ubuntu16.04LTS

s-vijayaselvan, 2021-06-26:
when i try to install pybombs .i got this. any one help me out.
sudo pip install pybombs

Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
  File "/home/.local/lib/python2.7/site-packages/pip/__init__.py", line 1, in <module>
    from typing import List, Optional
ImportError: No module named typing

s-vijayaselvan, 2021-06-26:
sudo pip3 install pybombs

 Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/local/lib/python3.5/dist-packages/setuptools/__init__.py", line 18, in <module>
        from setuptools.dist import Distribution
      File "/usr/local/lib/python3.5/dist-packages/setuptools/dist.py", line 585
        license_files: Optional[List[str]] = self.metadata.license_files
                     ^
    SyntaxError: invalid syntax
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-l46rd5b7/future/

willcode, 2021-06-26:
On an `ubuntu:16.04` container, ran
```
apt update
apt install python3-pip
pip3 install pybombs
```
with no errors.

argilo, 2022-12-15:
Closing, since pip installation apparently works on Ubuntu 16.04.

Also, GNU Radio now requires Ubuntu >= 20.04, so I wouldn't recommend trying to use PyBOMBS on Ubuntu 16.04.

closed 2022-12-15 by argilo