gnuradio/pybombs #599: USRP - N300 RunTimeError

nielsvanetten, 2021-06-29:
Dear all,

I'm trying to run the Ettus N300 on GNU radio.
When I'm running GNU radio via: `pybombs run gnuradio-companion`, I get the following error when trying to run the USRP Source block:
```
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-62-g7a3f1516
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.10.2,type=n3xx,product=n300,serial=320C34D,claimed=False,addr=192.168.10.2
Traceback (most recent call last):
  File "/home/niels/top_block.py", line 187, in <module>
    main()
  File "/home/niels/top_block.py", line 163, in main
    tb = top_block_cls()
  File "/home/niels/top_block.py", line 84, in __init__
    channels=list(range(0,1)),
  File "/home/niels/prefix-3.8/lib/python3/dist-packages/gnuradio/uhd/__init__.py", line 125, in constructor_interceptor
    return old_constructor(*args)
  File "/home/niels/prefix-3.8/lib/python3/dist-packages/gnuradio/uhd/uhd_swig.py", line 3298, in make
    return _uhd_swig.usrp_source_make(device_addr, stream_args, issue_stream_cmd_on_start)
RuntimeError: rpc::timeout: Timeout of 2000ms while calling RPC function 'get_num_xbars'
```
Note that `uhd_usrp_probe` works without warnings and also the UHD example: 
`/usr/local/lib/uhd/examples/rx_ascii_art_dft --args "master_clock_rate=125e6,mgmt_addr=192.168.1.151,addr=192.168.10.2" --freq 98.5e6 --rate 2.5e6 --gain 50 --ref-lvl="-50" --dyn-rng 90 --ant "RX2" --subdev "A:0"`, runs like a charm.

Thanks in advance

dkozel, 2021-06-29:
Hi @nielsvanetten
The error you're seeing is coming from UHD or gr-uhd and is highly unlikely to be related to running it through PyBOMBS. You'll have to ask NI about the problem. They use the usrp-users mailing list or support@ettus.com.
https://kb.ettus.com/Technical_Support

It is possible that there's an issue in gr-uhd, in that case the best place to ask is either live at https://chat.gnuradio.org or on the mailing list.
https://lists.gnu.org/mailman/listinfo/discuss-gnuradio

If you or someone from NI can identify a problem in the GNU Radio code then we can setup an issue on the main GNU Radio GitHub repo and get it fixed.

Cheers!

closed 2021-06-29 by dkozel

nielsvanetten, 2021-06-29:
Thanks for your support