gnuradio/pybombs #602: Downloading recipes not possible with Python 3.9

dominformant, 2021-08-11:
Using Python 3.9, pybombs always crashes on `pybombs recipes add-defaults` with the following error:
```
Traceback (most recent call last):
  File "/home/mniestroj/.local/bin/pybombs", line 8, in <module>
    sys.exit(main())
  File "/home/mniestroj/.local/lib/python3.9/site-packages/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/home/mniestroj/.local/lib/python3.9/site-packages/pybombs/commands/base.py", line 195, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/home/mniestroj/.local/lib/python3.9/site-packages/pybombs/commands/base.py", line 120, in run
    return self.subcommands[self.args.sub_command]['run'](self)()
  File "/home/mniestroj/.local/lib/python3.9/site-packages/pybombs/commands/recipes.py", line 180, in run_add_defaults
    if not self.add_recipe_dir(alias, uri):
  File "/home/mniestroj/.local/lib/python3.9/site-packages/pybombs/commands/recipes.py", line 347, in add_recipe_dir
    Fetcher().fetch_url(uri, recipe_cache_top_level, alias, {}) # No args
  File "/home/mniestroj/.local/lib/python3.9/site-packages/pybombs/fetcher.py", line 73, in fetch_url
    result = fetcher.fetch_url(url, dest, dirname, args)
  File "/home/mniestroj/.local/lib/python3.9/site-packages/pybombs/fetchers/git.py", line 102, in fetch_url
    subproc.monitor_process(
  File "/home/mniestroj/.local/lib/python3.9/site-packages/pybombs/utils/subproc.py", line 261, in monitor_process
    raise ex
  File "/home/mniestroj/.local/lib/python3.9/site-packages/pybombs/utils/subproc.py", line 243, in monitor_process
    while monitor_thread.isAlive:
AttributeError: 'Thread' object has no attribute 'isAlive'
```
It seems like python removed the `isAlive` attribute in one of the recent releases. On another machine with python 3.6 everything works fine.

willcode, 2021-08-11:
This was fixed at the end of last year, but the latest tagged version (2.3.4) does not include the fix. If you are able, install it the latest `master` branch, or make the following change (`isAlive` -> `is_alive()`):

```
diff --git a/pybombs/utils/subproc.py b/pybombs/utils/subproc.py
index c372e3b..c46d767 100644
--- a/pybombs/utils/subproc.py
+++ b/pybombs/utils/subproc.py
@@ -240,7 +240,7 @@ def monitor_process(args, **kwargs):
             args=(quit_event, args, kwargs)
         )
         monitor_thread.start()
-        while monitor_thread.isAlive:
+        while monitor_thread.is_alive():
             # Try if it's finished:
             monitor_thread.join(1)
             if quit_event.is_set() or not monitor_thread.is_alive():
```

dominformant, 2021-08-12:
Thanks for the reply, I looked back some issues and merge requests, but obviously not far enough...

closed 2021-08-12 by dominformant