gnuradio/pybombs #614: Error installing package libvolk

capsulecorplab, 2022-04-21:
Reinstalled `pybombs` from source (cloning repo and running `python setup.py build`) in a virtual environment, using apt installed `python3.8-venv`, then ran `pybombs auto-config`, followed by `pybombs recipes add-defaults` without issues. Encountered error installing package `libvolk` after running `pybombs prefix init ~/gr38 -R gnuradio-default`

```
$ pybombs prefix init ~/gr38 -R gnuradio-default
[INFO] Prefix Python version is: 3.8.10
[INFO] PyBOMBS Version 2.3.5
[WARNING] There already is a prefix in `/home/captainpike/gr38'.
Continue using this path Y/[N]? Y
[INFO] Prefix Python version is: 3.8.10
[INFO] Prefix Python version is: 3.8.10
[INFO] Installing default packages for prefix...
[INFO] 
  - gnuradio
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
   |
   +- uhd
   |
   \- libvolk
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: libvolk
[WARNING] A source build for package libvolk was requested, but binary install was found!
Install libvolk from source despite binary install available Y/[N]? Y
[INFO] Install python-apt to speed up apt processing.
warning: no common commits
remote: Enumerating objects: 10411, done.
remote: Counting objects: 100% (829/829), done.
remote: Compressing objects: 100% (364/364), done.
remote: Total 10411 (delta 462), reused 772 (delta 440), pack-reused 9582
Receiving objects: 100% (10411/10411), 3.04 MiB | 2.64 MiB/s, done.
Resolving deltas: 100% (7051/7051), done.
From https://github.com/gnuradio/volk
 * [new branch]          1.4-support        -> libvolk/1.4-support
 * [new branch]          apply_PR_#155      -> libvolk/apply_PR_#155
 * [new branch]          balister-patch-1   -> libvolk/balister-patch-1
 * [new branch]          bhilburn-volk-lgpl -> libvolk/bhilburn-volk-lgpl
 * [new branch]          main               -> libvolk/main
 * [new branch]          maint              -> libvolk/maint
 * [new branch]          master             -> libvolk/master
 * [new tag]             v1.3.1             -> v1.3.1
 * [new tag]             v1.0               -> v1.0
 * [new tag]             v1.0.1             -> v1.0.1
 * [new tag]             v1.0.2             -> v1.0.2
 * [new tag]             v1.1               -> v1.1
 * [new tag]             v1.1.1             -> v1.1.1
 * [new tag]             v1.1.2             -> v1.1.2
 * [new tag]             v1.2               -> v1.2
 * [new tag]             v1.2.1             -> v1.2.1
 * [new tag]             v1.2.2             -> v1.2.2
 * [new tag]             v1.2.3             -> v1.2.3
 * [new tag]             v1.3               -> v1.3
 * [new tag]             v1.4               -> v1.4
 * [new tag]             v2.0.0             -> v2.0.0
 * [new tag]             v2.1.0             -> v2.1.0
 * [new tag]             v2.2.0             -> v2.2.0
 * [new tag]             v2.2.1             -> v2.2.1
 * [new tag]             v2.3.0             -> v2.3.0
 * [new tag]             v2.4.0             -> v2.4.0
 * [new tag]             v2.4.1             -> v2.4.1
 * [new tag]             v2.5.0             -> v2.5.0
 * [new tag]             v2.5.1             -> v2.5.1
Cloning into 'libvolk'...
Submodule 'cpu_features' (https://github.com/google/cpu_features.git) registered for path 'cpu_features'
submodule 'cpu_features' cannot add alternate: path '/home/captainpike/.pybombs/gitcache/modules/cpu_features/' does not exist
Cloning into '/home/captainpike/gr38/src/libvolk/cpu_features'...
remote: Enumerating objects: 1631, done.        
remote: Counting objects: 100% (476/476), done.        
remote: Compressing objects: 100% (139/139), done.        
remote: Total 1631 (delta 396), reused 337 (delta 337), pack-reused 1155        
Receiving objects: 100% (1631/1631), 483.74 KiB | 1.04 MiB/s, done.
Resolving deltas: 100% (1077/1077), done.
Submodule path 'cpu_features': checked out '188d0d3c383689cdb6bb70dc6da2469faec84f61'
[INFO] Install python-apt to speed up apt processing.
Configuring: (100%) [=======================================================================================================================]
[WARNING] Configuration failed. Re-trying with higher verbosity.
-- Build type set to RelWithDebInfo.
-- Extracting version information from git describe...
-- 
-- Python checking for python >= 3.4
-- Python checking for python >= 3.4 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - not found
CMake Error at CMakeLists.txt:174 (message):
  Mako templates required to build VOLK

-- Configuring incomplete, errors occurred!
See also "/home/captainpike/gr38/src/libvolk/build/CMakeFiles/CMakeOutput.log".
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package libvolk:
Configuration failed
[ERROR] Error installing package libvolk. Aborting.
```

capsulecorplab, 2022-05-02:
Ran a `pip install mako requests numpy`, followed by `pybombs prefix init ~/gr38 -R gnuradio38` in the same virtual environment, but still running into a libvolk error in same virtual environment

```
[INFO] Prefix Python version is: 3.8.10
[INFO] PyBOMBS Version 2.3.5
[WARNING] There already is a prefix in `/home/captainpike/gr38'.
Continue using this path Y/[N]? Y
[INFO] Prefix Python version is: 3.8.10
[INFO] Prefix Python version is: 3.8.10
[INFO] Installing default packages for prefix...
[INFO] 
  - gnuradio
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
   |
   +- libvolk
   |
   +- soapysdr
   |
   \- uhd
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: uhd
[INFO] Install python-apt to speed up apt processing.
remote: Enumerating objects: 191, done.
remote: Counting objects: 100% (148/148), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 191 (delta 138), reused 148 (delta 138), pack-reused 43
Receiving objects: 100% (191/191), 65.45 KiB | 515.00 KiB/s, done.
Resolving deltas: 100% (147/147), completed with 69 local objects.
From https://github.com/EttusResearch/uhd
   197cdc4f6..04d14cd76  UHD-4.2    -> uhd/UHD-4.2
   197cdc4f6..5333d3d12  master     -> uhd/master
 * [new tag]             v4.2.0.0   -> v4.2.0.0
Cloning into 'uhd'...
remote: Enumerating objects: 1286, done.
remote: Counting objects: 100% (693/693), done.
remote: Total 1286 (delta 693), reused 693 (delta 693), pack-reused 593
Receiving objects: 100% (1286/1286), 1.15 MiB | 1.49 MiB/s, done.
Resolving deltas: 100% (896/896), completed with 210 local objects.
[INFO] Install python-apt to speed up apt processing.
Configuring: (100%) [=======================================================================================================================]
Building:    (100%) [=======================================================================================================================]
Installing:  (100%) [=======================================================================================================================]
[INFO] Installation successful.
[INFO] Installing package: soapysdr
warning: no common commits
remote: Enumerating objects: 10065, done.
remote: Counting objects: 100% (2026/2026), done.
remote: Compressing objects: 100% (657/657), done.
remote: Total 10065 (delta 1340), reused 1978 (delta 1325), pack-reused 8039
Receiving objects: 100% (10065/10065), 4.76 MiB | 2.41 MiB/s, done.
Resolving deltas: 100% (7710/7710), done.
From https://github.com/pothosware/SoapySDR
 * [new branch]          arg_info_work2  -> soapysdr/arg_info_work2
 * [new branch]          convert_work    -> soapysdr/convert_work
 * [new branch]          feat-py3find    -> soapysdr/feat-py3find
 * [new branch]          fix-windows-ci  -> soapysdr/fix-windows-ci
 * [new branch]          fix_travis      -> soapysdr/fix_travis
 * [new branch]          get-specific-setting-info -> soapysdr/get-specific-setting-info
 * [new branch]          gh-pages        -> soapysdr/gh-pages
 * [new branch]          luajit-support  -> soapysdr/luajit-support
 * [new branch]          maint           -> soapysdr/maint
 * [new branch]          master          -> soapysdr/master
 * [new branch]          swig-python-shadow-warning -> soapysdr/swig-python-shadow-warning
 * [new branch]          swig-reorganize -> soapysdr/swig-reorganize
 * [new branch]          time_work       -> soapysdr/time_work
[... 28 lines cut ...]
 * [new tag]             soapy-sdr-0.8.1 -> soapy-sdr-0.8.1
Cloning into 'soapysdr'...
Configuring: (100%) [=======================================================================================================================]
Building:    (100%) [=======================================================================================================================]
Installing:  (100%) [=======================================================================================================================]
[INFO] Installation successful.
[INFO] Installing package: libvolk
[WARNING] A source build for package libvolk was requested, but binary install was found!
Install libvolk from source despite binary install available Y/[N]? Y
`\Y' is not a valid response.
Install libvolk from source despite binary install available Y/[N]? Y
[WARNING] Build dir already exists: /home/captainpike/gr38/src/libvolk/build
Building:    (100%) [=======================================================================================================================]
[ 72%] Built target volk_obj
[ 74%] Linking C shared library libvolk.so
/usr/bin/ld: cannot find -lcpu_features
collect2: error: ld returned 1 exit status
make[2]: *** [lib/CMakeFiles/volk.dir/build.make:105: lib/libvolk.so.2.5.1] Error 1
make[1]: *** [CMakeFiles/Makefile2:227: lib/CMakeFiles/volk.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package libvolk:
Build failed.
[ERROR] Error installing package libvolk. Aborting.
```

garengllc, 2022-05-03:
I am having the same issue.  I rebuilt a Docker I had that worked a few weeks ago and now it won't get past libvolk.  I tried forcing the libvolk to v2.0.0 because that was a known good version, but that did not resolve the problem (makes me think pybombs as opposed to libvolk).  The command I had used was: `pip3 install --upgrade git+https://github.com/gnuradio/pybombs.git`

tecandrew, 2022-07-01:
still doesn't work. running into the same issue within a `pyenv` installation.

mcdiarmid, 2022-07-08:
Might not be relevant to what everyone here is experiencing -  I encountered this issue while I tried to init a new pybombs 3.9 and 3.10 prefix.  However, I had activated my existing pybombs3.8 prefix environment ` source ~/gnuradio38/setup_env.sh`.  After trying again ensuring this environment wasn't activated  `pybombs prefix init ~/gnuradio3.10.3.0 -R gnuradio310` was successful.

garengllc, 2022-07-12:
Per a recommendation someone else gave me, I am using a particular commit of the recipes now
`
pybombs -y -p $PREFIX recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git@15157e7
pybombs -y -p $PREFIX recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git@506e0f5
`
That seems to do the trick.

aardric, 2022-07-15:
I have a similar problem with libvolk while executing recipe gnuradio39:
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: cannot find -lcpu_features
collect2: error: ld returned 1 exit status
Rather than try to coax gr-recipes.git@15157e7 into working I would like to know what is wrong with the latest recipe repo. I have been using PyBOMBS for almost two years now and prefer not to give it up without a fight. I look at this as a learning experience.

rsmegal, 2022-07-16:
After inspecting the file "prefix/gr39/src/libvolk/build/CMakeCache.txt" I noticed that CpuFeatures_DIR:PATH was given the value pertaining to a local installation of volk (~/.local/lib64). I renamed this directory with the consequence that "CpuFeatures_DIR:PATH=CpuFeatures_DIR-NOTFOUND."  Installing gnuradio with "pybombs -vv -p gr39 install gnuradio" was successful. This seems like a kludge so if I wanted to keep the local installation of volk, what would I do? Forging ahead, I will uninstall the local installation of volk.

GentlemanSch, 2023-09-08:
I fixed this by installing an older version of mako ( 0.8.0 ) before the installer, running:

`python3 -m pip install mako==0.8.0`

worked for me. It seems that it isn't checking the major version, just the minor.