gnuradio/pybombs #620: installation problem of gr-ieee-80211, version 3.9

franckstik, 2022-09-22:
Hello,
when I try to install gr-ieee-80211 on Ubuntu 20.04, with pybombs, with gnu radio version 3.9, I get the following message error.
*************
Clonage dans 'gr-ieee802-11'...
remote: Enumerating objects: 2773, done.
remote: Counting objects: 100% (29/29), done.
remote: Compressing objects: 100% (22/22), done.
remote: Total 2773 (delta 11), reused 14 (delta 7), pack-reused 2744
Réception d'objets: 100% (2773/2773), 1.03 Mio | 2.88 Mio/s, fait.
Résolution des deltas: 100% (1908/1908), fait.
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   Found gmp, version 6.2.0
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Using GMP.
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
-- Found Volk: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.10") 
-- Checking for module 'sndfile'
--   Found sndfile, version 1.0.28
-- Found SNDFILE: /usr/lib/x86_64-linux-gnu/libsndfile.so  
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.8
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Looking for xmmintrin.h
-- Looking for xmmintrin.h - found
-- Using install prefix: /usr/local
-- Building for version: 86965650 / 1.1.0git
-- Found SWIG: /usr/bin/swig4.0 (found version "4.0.1") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") 
CMake Error at swig/CMakeLists.txt:34 (include):
  include could not find load file:

    GrSwig

CMake Error at swig/CMakeLists.txt:48 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

-- Configuring incomplete, errors occurred!
See also "/home/tim/prefix-3.9/src/gr-ieee802-11/build/CMakeFiles/CMakeOutput.log".
See also "/home/tim/prefix-3.9/src/gr-ieee802-11/build/CMakeFiles/CMakeError.log".
*************
Would you have the possibility to correct the problem ?
Thank you

willcode, 2022-09-22:
The default branch for gr-ieee-80211 is maint-3.8. Create (or edit) a `.pybombs/config.yml` file in the prefix directory and include
```
- packages:
    gr-ieee-80211:
      gitbranch: maint-3.9
```
For reference, here's the entire file in my 3.9 prefix. Some of the stuff in here is outdated, but you get this idea:
```
!!omap
- categories:
    hardware:
      forcebuild: true
    common:
      forcebuild: true
- packages:
    gnuradio:
      forcebuild: true
      gitbranch: maint-3.9
    gqrx:
      forcebuild: true
    uhd:
      gitbranch: UHD-4.1
    gr-xcorrelate:
      gitbranch: gr39
    gr-foo:
      gitbranch: maint-3.9
    gr-ieee-80211:
      gitbranch: maint-3.9
    gr-ieee-802154:
      gitbranch: maint-3.9
    gr-rds:
      gitbranch: maint-3.9
```

willcode, 2022-09-22:
Closing this issue, since this is a help request. Please use chat.gnuradio.org if you need more help.

closed 2022-09-22 by willcode