gnuradio/pybombs #621: Pybombs failed to build uhd

alexis084, 2022-11-09:
I'm trying to install gnuradio and uhd on a raspberry pi 3 model b+. It is a 32-bit armv7l system running Raspberry Pi OS. `pybombs prefix init ~/prefix-3.10 -R gnuradio-default` fails when trying to build uhd saying:
```
In file included from /home/alexus084/prefix-3.10/src/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h: In member function ‘virtual void __convert_fc32_1_sc16_item32_le_1_PRIORITY_SIMD::operator()(const input_type&, const output_type&, size_t)’:
/usr/lib/gcc/arm-linux-gnueabihf/10/include/arm_neon.h:6744:1: error: inlining failed in call to ‘always_inline’ ‘float32x4_t vdupq_n_f32(float32_t)’: target specific option mismatch
 6744 | vdupq_n_f32 (float32_t __a)
      | ^~~~~~~~~~~
```
The same error happens for multiple functions across the files mentioned in the above error message. I've attached the full terminal output along with the commands I ran below.
I'm hesitant to change these files because it seems that any changes could have unintended consequences on the system. 
Has anyone ran into this issue before? Any suggestions would be greatly appreciated.

[pybombs_uhd_error.txt](https://github.com/gnuradio/pybombs/files/9975203/pybombs_uhd_error.txt)

argilo, 2022-11-09:
I suspect you've run into this issue: https://github.com/EttusResearch/uhd/issues/350

If that's the case, you might be able to resolve this by editing `~/.pybombs/recipes/gr-recipes/uhd.lwr` and adding something like:
```
vars:
  config_opt: ' -DCMAKE_CXX_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a53 -Wno-psabi" -DCMAKE_C_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a53 -Wno-psabi" -DCMAKE_ASM_FLAGS:STRING="-march=armv7-a -mfloat-abi=hard -mfpu=neon -mtune=cortex-a53'
```

argilo, 2022-11-10:
You could rebuild UHD by running `pybombs rebuild uhd`. But it may be easier to just start over from scratch with a new prefix, so that the dependent packages are built as well.

alexis084, 2022-11-10:
I made the suggested change, deleted old prefixes and now get this output:
```
alexus084@raspberrypi:~ $ pybombs prefix init ~/prefix-3.10 -R gnuradio-default
[INFO] Prefix Python version is: 3.9.2
[INFO] PyBOMBS Version 2.3.5
[INFO] Creating directory `/home/alexus084/prefix-3.10'
[INFO] Prefix Python version is: 3.9.2
[INFO] Creating new config file /home/alexus084/prefix-3.10/.pybombs/config.yml
[INFO] Prefix Python version is: 3.9.2
[INFO] Installing default packages for prefix...
[INFO]
  - gnuradio
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
   |
   +- soapysdr
   |
   +- libvolk
   |
   +- uhd
   |
   \- libiio
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: libiio
Cloning into 'libiio'...
remote: Enumerating objects: 22, done.
remote: Counting objects: 100% (11/11), done.
remote: Total 22 (delta 11), reused 11 (delta 11), pack-reused 11
Receiving objects: 100% (22/22), 18.77 KiB | 2.35 MiB/s, done.
Resolving deltas: 100% (11/11), completed with 6 local objects.
Note: switching to 'tags/v0.21'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 565bf68ec Bump to version v0.21
Configuring: (100%) [========================================================================================================================================]
Building:    (100%) [========================================================================================================================================]
Installing:  (100%) [========================================================================================================================================]
[INFO] Installation successful.
[INFO] Installing package: uhd
[WARNING] A source build for package uhd was requested, but binary install was found!
Install uhd from source despite binary install available Y/[N]? Y
Cloning into 'uhd'...
remote: Enumerating objects: 1351, done.
remote: Counting objects: 100% (741/741), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 1351 (delta 738), reused 738 (delta 738), pack-reused 610
Receiving objects: 100% (1351/1351), 1.16 MiB | 2.19 MiB/s, done.
Resolving deltas: 100% (941/941), completed with 223 local objects.
Updating files: 100% (6768/6768), done.
Traceback (most recent call last):
  File "/usr/local/bin/pybombs", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.9/dist-packages/pybombs/main.py", line 32, in main
    return dispatch() or 0
  File "/usr/local/lib/python3.9/dist-packages/pybombs/commands/base.py", line 195, in dispatch
    return get_cmd_dict(cmd_list)[args.command](cmd=args.command, args=args).run()
  File "/usr/local/lib/python3.9/dist-packages/pybombs/commands/base.py", line 120, in run
    return self.subcommands[self.args.sub_command]['run'](self)()
  File "/usr/local/lib/python3.9/dist-packages/pybombs/commands/prefix.py", line 178, in run_init
    if not self._init_prefix(
  File "/usr/local/lib/python3.9/dist-packages/pybombs/commands/prefix.py", line 338, in _init_prefix
    return install_dependencies(prefix_recipe.depends)
  File "/usr/local/lib/python3.9/dist-packages/pybombs/commands/prefix.py", line 314, in install_dependencies
    return install_manager.InstallManager().install(
  File "/usr/local/lib/python3.9/dist-packages/pybombs/install_manager.py", line 138, in install
    if not self.pm.install(pkg, install_type="source", static=static, verify=verify):
  File "/usr/local/lib/python3.9/dist-packages/pybombs/package_manager.py", line 237, in install
    install_result = self._std_package_operation(
  File "/usr/local/lib/python3.9/dist-packages/pybombs/package_manager.py", line 283, in _std_package_operation
    result = getattr(pkgr, operation)(rec, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/pybombs/packagers/source.py", line 105, in install
    self.run_build(
  File "/usr/local/lib/python3.9/dist-packages/pybombs/packagers/source.py", line 235, in run_build
    recipe.vars['builddir'] = builddir
TypeError: 'NoneType' object does not support item assignment
alexus084@raspberrypi:~ $ 
```
Could the issue be related to installing uhd from source despite having a binary install?
Should I make this a new issue?

argilo, 2022-11-10:
I suspect the problem is that your UHD recipe (uhd.lwr) is broken.