gnuradio/pybombs #624: could not find GrSwig.

hfarooq98, 2022-12-14:
Hi, I installed pybombs using:

sudo apt-get install python3-pip
sudo pip3 install pybombs
pybombs auto-config
pybombs recipes add-defaults
pybombs prefix init ~/prefix-3.10 -R gnuradio-default
source ~/prefix-3.10/setup_env.sh
gnuradio-companion

now when I cmake [https://github.com/kit-cel/gr-radar](url) this repo it shows:

CMake Error at swig/CMakeLists.txt:37 (include):
  include could not find requested file:
    GrSwig
CMake Error at swig/CMakeLists.txt:50 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".

I found that it works with gnu radio maint-3.8 and I installed 3.10 so how can I go back to 3.8 without installing whole pybombs again?

willcode, 2022-12-14:
Multiple pybombs prefixes can exist on the same machine, and you'll need to install the 3.8 version from scratch (starting with the `pybombs prefix init` line. There is a `gnuradio38` target.

closed 2022-12-14 by willcode

hfarooq98, 2022-12-15:
sudo apt-get install python3-pip
sudo pip3 install pybombs
pybombs auto-config
pybombs recipes add-defaults
pybombs prefix init ~/prefix-gnuradio38 -R gnuradio-default
source ~/prefix-gnuradio38 setup_env.sh
gnuradio-companion

will this install 3.8?

hfarooq98, 2022-12-15:
huzaifa@huzaifPC:~$ pybombs prefix init ~/prefix-3.8 -R gunradio38
[INFO] Prefix Python version is: 3.10.6
[INFO] PyBOMBS Version 2.3.5
[ERROR] Could not find recipe for `gunradio38'

Why is this error popping up?

willcode, 2022-12-15:
Spelling