gnuradio/pybombs #628: uhd problem

Eetuheino03, 2023-04-14:
pybombs rebuild uhd                                                 
[INFO] Prefix Python version is: 3.11.2
[INFO] PyBOMBS Version 2.3.5
[INFO] Rebuilding package: uhd
[INFO] Removing old build directory.
Configuring: (100%) [=====================================================================================================================================]
Building:    (100%) [=====================================================================================================================================]
[  2%] Built target uhd_rpclib
[  4%] Built target uhd-resources
[  4%] Building CXX object lib/CMakeFiles/uhd.dir/usrp/mpmd/sim_find.cpp.o
In file included from /home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/attr.h:11,
                 from /home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:43,
                 from /home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/embed.h:11,
                 from /home/kali/pybombs/bladeRF/src/uhd/host/lib/usrp/mpmd/sim_find.cpp:14:
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h: In function ‘std::string pybind11::detail::error_string()’:
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:422:36: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  422 |                 "  " + handle(frame->f_code->co_filename).cast<std::string>() +
      |                                    ^~
In file included from /usr/include/python3.11/Python.h:42,
                 from /home/kali/pybombs/bladeRF/src/uhd/host/lib/usrp/mpmd/sim_find.cpp:13:
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:422:75: error: expected primary-expression before ‘>’ token
  422 |                 "  " + handle(frame->f_code->co_filename).cast<std::string>() +
      |                                                                           ^
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:422:77: error: expected primary-expression before ‘)’ token
  422 |                 "  " + handle(frame->f_code->co_filename).cast<std::string>() +
      |                                                                             ^
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:424:29: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  424 |                 handle(frame->f_code->co_name).cast<std::string>() + "\n";
      |                             ^~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:424:64: error: expected primary-expression before ‘>’ token
  424 |                 handle(frame->f_code->co_name).cast<std::string>() + "\n";
      |                                                                ^
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:424:66: error: expected primary-expression before ‘)’ token
  424 |                 handle(frame->f_code->co_name).cast<std::string>() + "\n";
      |                                                                  ^
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:425:26: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  425 |             frame = frame->f_back;
      |                          ^~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h: In function ‘pybind11::function pybind11::detail::get_type_override(const void*, const type_info*, const char*)’:
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:2119:49: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘frame’; did you mean ‘cframe’?
 2119 |     PyFrameObject *frame = PyThreadState_Get()->frame;
      |                                                 ^~~~~
      |                                                 cframe
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:2120:41: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 2120 |     if (frame && (std::string) str(frame->f_code->co_name) == name &&
      |                                         ^~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:2121:14: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 2121 |         frame->f_code->co_argcount > 0) {
      |              ^~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:2124:18: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 2124 |             frame->f_locals, PyTuple_GET_ITEM(frame->f_code->co_varnames, 0));
      |                  ^~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
In file included from /usr/include/python3.11/Python.h:38:
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pytruct _frame’}
 2124 |             frame->f_locals, PyTuple_GET_ITEM(frame->f_code->co_varna
      |                                                    ^~
/usr/include/python3.11/pyport.h:24:38: note: in definition of macro ‘_Py_CAS
   24 | #define _Py_CAST(type, expr) ((type)(expr))
      |                                      ^~~~
/usr/include/python3.11/cpython/tupleobject.h:30:38: note: in expansion of ma
   30 | #define PyTuple_GET_ITEM(op, index) (_PyTuple_CAST(op)->ob_item[index
      |                                      ^~~~~~~~~~~~~
/home/kali/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/py
 2124 |             frame->f_locals, PyTuple_GET_ITEM(frame->f_code->co_varna
      |                              ^~~~~~~~~~~~~~~~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyF
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
make[2]: *** [lib/CMakeFiles/uhd.dir/build.make:2973: lib/CMakeFiles/uhd.dir/
make[1]: *** [CMakeFiles/Makefile2:1084: lib/CMakeFiles/uhd.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package uhd:
[ERROR] Build failed.
[ERROR] Error rebuilding package uhd. Aborting.

Eetuheino03, 2023-04-14:
OS: Kali Linux Latest
I have been installed all requirements but it's just not working.
I have Bladerf 2.0 A4

willcode, 2023-06-15:
This is likely a version mismatch between some autogenerated pybind11 code and the pybind11 library. I also do not see `PyFrameObject` in UHD.

In any case, PyBOMBS is just the building - it does not have anything to do with the UHD or other code. If you haven't already, try this from scratch on a clean system to try to rule out version mismatches.

closed 2023-06-15 by willcode

fabiozerbetto, 2023-06-16:
> OS: Kali Linux Latest I have been installed all requirements but it's just not working. I have Bladerf 2.0 A4

I was able to build gnuradio using pybombs by overriding the building of uhd through pybombs configuration.
In my case I had just installed uhd libs with precompiled package using apt in debian, than i edit the
~/.pybombs/config.yml by adding following lines:

```
- packages:
    uhd:
        forebuild: False
```

It worked to me.